srsran/srsRAN_4G #1061: EPC support for GTPV2

getabc, 2022-12-16:
Setting up a connection between Open5GS and SRSENB and unable to get a data connection

SRSENB IP 10.10.2.16
Open5GS IP 10.10.2.17

Open5GS MME log reports a connection error to ENB
[gtp] WARNING: [2] LOCAL  No Reponse. Give up! for step 1 type 32 peer [10.10.2.16]:2123

There is no process listening to UDP 2123 on the SRSENB

I found mention of the ports in the source code.
#define OGS_GTPV1_U_UDP_PORT            2152
#define OGS_GTPV2_C_UDP_PORT            2123

Should the SRSENB be listening to 2123?

iman432, 2022-12-20:
Hi,

MME doesn't connect by GTPv2 to eNB, it connects by S1AP,

eNB has tunnel by SGW only and its GTPv1.

getabc, 2022-12-21:
That makes way more sense - thank you.

closed 2022-12-21 by getabc