srsran/srsRAN_4G #1062: Failed to read rb configuration while running srsenb

FaribaNajm, 2022-12-16:
I have installed the BladeRF driver from source through this link https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux
Then I have installed srsRAN from source through  https://docs.srsran.com/en/latest/general/source/1_installation.html
Right now when I want to run the srsenb, it says it cannot open rb.conf. I have attached the commands I tried and no success. Just one thing that I do not have hardware Blade RF connected. 
I would appreciate if some one could help me with this issue. 
root@fariba:~# srsenb
Active RF plugins: libsrsran_rf_soapy.so libsrsran_rf_blade.so librsran_rf_zmq.
SO
Inactive RF plugins:
Software Radio Systems LTE eNodeB

Couldn't open , trying /root/.config/srsran/enb.conf
Couldn't open /root/.config/srsran/enb.conf either, trying /etc/srsran/enb.conf
Failed to read eNB configuration file /etc/srsran/enb.conf exiting

fariba@fariba:/SRSRAN/srsenb$ sudo srsenb ~/.config/srsran/enb.conf 

Active RF plugins: libsrsran_rf_soapy.so libsrsran_rf_blade.so libsrsran_rf_zmq.
SO
Inactive RF plugins:
Software Radio Systems LTE eNodeB
Reading configuration file /home/fariba/.config/srsran/enb.conf
Couldn't open rb.conf, trying /root/.config/srsran/rb.conf
Couldn't open /root/.config/srsran/rb.conf either, trying /etc/srsran/rb.conf
Failed to read RB configuration file /etc/srsran/rb.conf
- exiting

iman432, 2022-12-20:
Hi,

First you should find location of rb.conf then specify at enb.conf , based on log its trying /root/.config/srsran/rb.conf  and /etc/srsran/rb.conf but seems rb.conf isn't there.

[image]

FaribaNajm, 2022-12-20:
Hello,
Thanks for your reply. I can find the rb.conf in my home/srsRAN, however I don't know how to specify at enb.conf.
Thanks.

[Capture]<https://user-images.githubusercontent.com/44318450/208650900-d1aff7a5-fea1-422a-a843-6ea9dd4e2aac.PNG>