srsran/srsRAN_4G #1065: dynamic MCS adaptation

aabedii, 2022-12-19:
## Issue Description ##
[Describe the issue in detail]
Missing dynamic MCS adaptation for NR

## Setup Details ##
5G SA or NSA

## Expected Behavior ##
MCS changes dynamically as the channel changes

## Actual Behaviour ##
No support. MCS is fixed.

## Steps to reproduce the problem ##
Missing feature.

## Additional Information ##
Is there any plan to add this feature soon? If not, could you please refer me to how one can add this functionality in the code? Specifically, where in the code I can implement this feature.

andrepuschmann, 2022-12-20:
We will release a complete new gNB stack early next year which will support dynamic MCS selection. The gNB in the current srsRAN repo will not get this feature backported.

closed 2022-12-20 by andrepuschmann

teo-tsou, 2023-06-09:
Is this release already out? Also if I want to develop such functionality in the srsLTE, could you please suggest me which file should I modify on the source code? Thank you!

ismagom, 2023-06-10:
Hi @teo-tsou,

Yes the new release is here: https://github.com/srsran/srsRAN_Project

teo-tsou, 2023-06-10:
Thank you very much for your reply. Still can't find out how to change mcs dynamically, is there any doc? Can you please please help me on reaching to the cqi/mcs lookup table?

frankist, 2023-06-11:
You can find the files `mcs_calculator.h` / `mcs_calculator.cpp` in the `lib/scheduler/support` directory.