srsran/srsRAN_4G #1066: srsue is stopping at RRC release state

pksahoo2022, 2022-12-20:
## Issue Description ##
[Describe the issue in detail]
gnb is disconnecting after UE runs. UE is stopping at RRC released state. 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
5g  core hosted on ubuntu 20.04
srsRAN on ubuntu 18.04

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
[image]
[image]
[image]

Emmanuel2248, 2023-02-05:
i have the same problem as well. Did you get the solution to this problem

ridzafauzi, 2023-02-10:
there is an inactivity timer in srsgnb which is "inactivity_timeout_ms". you may increase it so that gnb wont release the ue quickly.

srsRAN/srsgnb/hdr/stack/rrc/rrc_nr_config.h:71:  uint32_t           inactivity_timeout_ms = 100000;

wmcclements, 2023-10-25:
I have a fix for this issue.

The problem is the srsUE should be sending an Odd digit indication (as the IMEI is 15 digits in length) but it's sending an Even digit indication.

Hence the core is not liking this and not accepting the Identity Response.

The following simple code change (and rebuild) will fix this:

~/srsRAN_4G$ git diff
diff --git a/srsue/src/stack/upper/nas_5g.cc b/srsue/src/stack/upper/nas_5g.cc
index a3ebdedac..1b55d37bc 100644
--- a/srsue/src/stack/upper/nas_5g.cc
+++ b/srsue/src/stack/upper/nas_5g.cc
@@ -724,6 +724,11 @@ int nas_5g::send_identity_response(srsran::nas_5g::identity_type_5gs_t::identity
     } break;
     case (identity_type_5gs_t::identity_types_::imei): {
       srsran::nas_5g::mobile_identity_5gs_t::imei_s& imei = identity_response.mobile_identity.set_imei();
+
+      // Fix for the Odd/Even Indicator IMEI Identity Request Issue
+      // IMEI is always 15-digits in length (odd) - hence set odd_even_indicator to true
+      imei.odd_even_indicator = true;
+
       usim->get_imei_vec(imei.imei.data(), 15);
     } break;
     case (identity_type_5gs_t::identity_types_::imeisv): {