srsran/srsRAN_4G #1068: Problem with certain COTS UE discovering network

tiger762, 2022-12-22:
Hello,

I have been successfully running srsenb for several years, with both srsepc as well as Open5GS EPC. Using either srsue or an LG K7 (2018 vintage) I can discover the network, get assigned an IP address and watch videos of big cats on Youtube with a decent 10Mbps downlink throughput.

The problem comes in when the CTO asks me to integrate Open5GS with Kamailio IMS. The LG K7 does not do VoLTE, so I tried a OnePlus model 5 but it cannot see the srsRAN network. Tried my personal cell (Samsung S8+, Sprint, unlocked) and it sees all the major wireless carriers (AT&T, Verizon, T-Mobile) but not mine. Have read that the IPhone 12 does VoLTE out of the box (no need to dork around with blasting firmware and other terrifying activities) so I bought an open-box IPhone 12. It can't discover the network either. I have tried all combinations of:

Band 1,2,4,7,12 (this is in the US, so all these bands should work)
PRB 6,15,25,50
as well as -2ppm, -1ppm, 0, 1ppm, 2ppm frequency offsets, all to no avail

I have opened another issue on the use of an external GPSDO in case it is related to frequency stability. I tend to think that the higher-end UE look at the SIB, see something they don't like, and keep scanning. I have included the config files I use for srsenb. I am eager to solve this, so any additional info you need will be provided. I just am at a loss of what other info would help.

Intel i5-7500k, Centos 8
UHD 3.15.0
srsRAN 21.4.0

One thing I will try to do is use srsue to look at the SIB that T-Mobile broadcasts. TMO is a carrier that the IPhone has no problems discovering, so maybe comparing their SIB with mine would be insightful(?)

Thank you in advance!
-KEF

[drb.txt](https://github.com/srsran/srsRAN/files/10283077/drb.txt)
[enb.txt](https://github.com/srsran/srsRAN/files/10283078/enb.txt)
[rr.txt](https://github.com/srsran/srsRAN/files/10283079/rr.txt)
[sib.txt](https://github.com/srsran/srsRAN/files/10283080/sib.txt)

tiger762, 2023-01-10:
So.... No one else has this problem?

I used a 2nd Ettus B200 to capture the MIB/SIB of srsENB in Wireshark, then captured the same from T-Mobile, Verizon, AT&T and Sprint to see if there was anything obviously missing. What we are up against is the chipset manufacturers (Broadcom, Intel, etc) have put in intelligence to avoid what appears to be an "IMSI catcher". The strategy is to send back to the UE a "identity can't be derived by network" when the UE does a TAU. This then makes the phone send the IMSI, at least that's the theory. In my case though, the IPhone does not even get that far.

Have used a NanoVNA (Ebay, $140) to make sure that the antenna I am using is reasonable for the 1.9GHz and 2.1GHz bands I am playing in. S11 reflection down below -20db. Also, using an external GPSDO (Ebay, $90) to get clock error around +/- 10Hz

I work for a wireless carrier in the US. I have a real USIM, with known Ki, OPc and IMSI.

If I manage to figure this out, is anyone interested in me posting my findings or are you all good?

cokia, 2023-01-22:
I'm having the same problem, and I'm getting a lot of help from your attempt. 
I will also share the debugging result asap to help solve this problem.

gmaruzz, 2023-02-27:
i'm interested too