srsran/srsRAN_4G #1072: COTS UE can not find the 5G SA network

Zerocl0, 2022-12-26:
## Issue Description ##
I follow the description in the [5G SA COTS UE](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html) section of the document, but I cannot find the Open5GS network from the mobile phone operator.

## Setup Details ##
Hardware:
AMD Ryzen 5 3550H

Operation System:
Ubuntu 20.04, running as a virtual machine on VMware

Configuration:
I used the configuration which is given in the 5G SA COTS UE of the document.

## Expected Behavior ##
The COTS UE can find the Open5GS Network and connect successfully.

## Actual Behaviour ##
The COTS UE can not find the Open5GS Network in the "WiFi & Network > SIM & network settings > SIM > Network operators".

## Steps to reproduce the problem ##
enb cmd line info:
```
sudo srsenb ~/Desktop/enb.conf 
[sudo] password for zerocl: 
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/zerocl/Desktop/enb.conf...
Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
Couldn't open rb.conf, trying /root/.config/srsran/rb.conf
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor
WARNING: Could not verify cpu2 scaling governor
WARNING: Could not verify cpu3 scaling governor

Built in Release mode using commit 254cc719a on branch master.

Opening 1 channels in RF device=default with args=default
Supported RF device list: UHD zmq file
Trying to open RF device 'UHD'
NG connection successful
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_3.15.0.0-74-ge35f66e8
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 2.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=1842.5 Mhz, DL_SSB=1842.05 Mhz (SSB-ARFCN=368410), UL=1747.5 MHz for cc_idx=0 nof_prb=52
^CStopping ..
Saving MAC PCAP (DLT=149) to /tmp/enb_mac.pcap
---  exiting  ---
```

enb log:
[enb.txt](https://github.com/srsran/srsRAN/files/10302902/enb.txt)

## Additional Information ##
I guess it may be a problem with the srsenb, because "Late", "Overflow" and "Underflow" appear repeatedly in the log. But I don't know how to solve this problem. 
As shown below, I tried some methods, but they didn't work.
```
zerocl@ubuntu:~$ echo "performance" | sudo tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor
tee: '/sys/devices/system/cpu/cpu*/cpufreq/scaling_governor': No such file or directory
performance

```
```
zerocl@ubuntu:~$ cpufreq-info 
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 1:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 2:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 3:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.

zerocl@ubuntu:~$ sudo cpufreq-set -g performance
Error setting new values. Common errors:
- Do you have proper administration rights? (super-user?)
- Is the governor you requested available and modprobed?
- Trying to set an invalid policy?
- Trying to set a specific frequency, but userspace governor is not available,
   for example because of hardware which cannot be set to a specific frequency
   or because the userspace governor isn't loaded?
```

iman432, 2022-12-27:
Hi,

For first step its better to try a commercial PLMN as most of phones are limited for 5G for all PLMNs,
second step, its better to run on bare metal not as VM and also your scaling_governor issue comes from running on VM. (Don't forget to use USB 3 too)

Zerocl0, 2022-12-29:
Hello,
Thanks for your reply. I will try as you said later.

helloTkk, 2023-01-13:
> Hello, Thanks for your reply. I will try as you said later.

Hi, I'm facing the same issue as you, did you figure it out? I'm also using the COTS UE listed in that note, but it cannot find the network.

JoseSecmotic, 2023-02-27:
I am also facing the same issue. The network cannot be seen by my Nord Oneplus. It's using sysmocomISIM-SJA2
This [guide](https://gist.github.com/mrlnc/01d6300f1904f154d969ff205136b753) about SUCI Concealment has been followed.
5G is activated in the COTS UE preferences and the PLMN is a commercial one.

I have succeeded implementing the same scenario for a 4G network smoothly, but changing to 5G its being an issue.

Did anyone came to a solution? @helloTkk @Zerocl0 @iman432 ?

Thanks in advance

ismagom, 2023-03-22:
Can you try with the newest release? https://github.com/srsran/srsRAN_Project

ayamoheddine, 2023-03-23:
Before when using bladeRF the UE was able to see the network but not connecting to it because of MAC failure (the sim card wasn't sysmo sim). However, I'm facing problems with it (FPGA and firmware flashing error) once I used an external power supply. So, now I'm using b210 mini where I'm facing a problem with UE not being able to see the network.

tiger762, 2023-04-20:
I am having this same issue as well.

IPhone 12 (A2172), upgraded to iOS 16.4.1, now the "5G on" menu item appears. When I got the phone it had 16.0.2 and there was no mention of 5G anywhere.

Tried n78 (ARFCN 633332, 3499.98 MHz, TDD), confirmed with a TinySA spectrum analyzer. Using a 3.5GHz directional antenna.

Went back to n2 (ARFCN 392000, 1960 MHz, FDD) with a pair of antennas that work perfectly with 4G at that frequency and also get nothing.

What is the secret to getting 5G discovered? Or another question would be: What readily available COTS UE is able to discover? Thanks..

iman432, 2023-04-20:
iPhone and Samsung phones have limitation for 5G for all PLMNs, especially
for SA, its suggested to use PLMN of a commercial SA network for your SIM
cards,maybe it solves the issue, its worth to say even supported bands in
iPhone varies PLMN by PLMN.

Best Regards
Iman Mohammadi

tiger762, 2023-04-20:
That's a good point. I work for PLMN 310690 and have a legit SIM card with known Ki and OPc. We are a roaming partner of T-Mobile, which I believe this IPhone was originally sold to but it is unlocked ("No SIM restrictions"). Besides, it works fine on 4G (srsENB)

Interesting that while typing this, in the console I just caught this failure:

[ERROR] [STREAMER] The receive packet handler caught a value exception.
ValueError: bad vrt header or packet fragment
[ERROR] [B200] Got a ctrl packet with unknown SID 805371888
Error: unhandled error in Rx metadata ERROR_CODE_BAD_PACKET.Error: exceeded maximum number of timed out transmissions.
Error: exceeded maximum number of timed out transmissions.
Error: exceeded maximum number of timed out transmissions.
Error: exceeded maximum number of timed out transmissions.
Error: exceeded maximum number of timed out transmissions.
Error: exceeded maximum number of timed out transmissions.

I hit Ctrl-C but am not able to restart 'gnb' until I reset the USB device. This might be something to do with it

tiger762, 2023-04-20:
Used the TinySA to scan 5G band n71 (FDD, 617-652 DL). That's T-Mobile's 15MHz spectrum on the left, and my 5MHz 'gnb' on the right (centered at 645MHz). It still is not discovered by the IPhone but it was worth a try....

[image]

andrepuschmann, 2023-04-21:
As has already pointed out above - please use the new repo https://github.com/srsran/srsRAN_Project for all 5G SA related work. The SA gNB in this repo will not be updated and will not receive any bug-fixes.

If you see issues with the new repo - please open a new bug report there. Thanks

closed 2023-04-21 by andrepuschmann

tiger762, 2023-04-21:
I already am running the latest

┌──(root💀kali)-[~]
└─# cd srsRAN_Project

┌──(root💀kali)-[~/srsRAN_Project]
└─# git pull
Already up to date.

┌──(root💀kali)-[~/srsRAN_Project]
└─#