srsran/srsRAN_4G #1075: COTS UE connect with eNB but not stable, SNR value not stable

helloTkk, 2022-12-31:
## Issue Description ##
Hello,
I’m using a HUAWEI P40 pro to connect with eNB(USRP B210+GPSDO+2 VERT 900 antennas). The UE connects successfully, however, It only connects to the cell for less than 1 minute, then the connection is lost.
I have tired to adjust tx/rx gain in config file, but the problem here still exists. After checking the eNB log, I found the SNR(signal noise ratio) value is decreasing over time which maybe the cause for the connection loss. Initially, the SNR value is good, but after 30 seconds, the SNR goes to a bad value and connection lost.
The enb.log was attached and my setup details was listed. Does anyone knows the reason why this could happen or have any idea to address this issue? Do I need to change a USRP like N310 to make it stable?
Any help I would appreciate!!!

## Setup Details ##
Operation System: Ubuntu 20.04
srsRAN 22.04
RF: USRP B210+GPSDO+2 VERT 900 antennas
UHD version: 3.15
UE: HUAWEI P40 pro
Card type: Sysmocom USIM(sysmoUSIM-SJS1)
[enb.txt](https://github.com/srsran/srsRAN/files/10326765/enb.txt)

## Expected Behavior ##
UE connect to 5G SA network and remain stable.

## Actual Behaviour ##
UE connects successfully, however, It only connects to the cell for less than 1 minute, then the connection is lost. The SNR value in enb.log is decreasing over time.

## Steps to reproduce the problem ##
enb.conf was attached.
[enb.log](https://github.com/srsran/srsRAN/files/10326761/enb.log)
[enb.log](https://github.com/srsran/srsRAN/files/10326762/enb.log)

## Additional Information ##
If anyone encounter the same issue, welcome to discuss with me, my email is: tukai2259@gmail.com , we can solve the problem together.

helloTkk, 2022-12-31:
> enb.txt is my configuration file.
enb.log is the log file which contains SNR value I mentioned.

lgs96, 2023-07-02:
@helloTkk Hi, did you find the solution to this problem? I guess that I'm dealing with the similar problem.