srsran/srsRAN_4G #1080: Failed to initiate SCTP socket. Attempting reconnection in 10 seconds when run srsenb

sj5298, 2023-01-08:
## Issue Description ##
Hi all,

I'm just starting out to establish an LTE network using srsRAN but I'm facing a problem when connecting eNB to the core network. 
I can't get my srsENB and srsEPC connected.
To my understanding, it is the SCTP socket connection problem.

The error in srsENB:
Failed to initiate SCTP socket. Attempting reconnection in 10 seconds

==== eNodeB started ===
Type <t> to view trace
Failed to initiate SCTP socket. Attempting reconnection in 10 seconds

[image]
[image]

## Setup Details ##
My setup was listed as follows:
1x Raspberry pi as eNB
1x laptop with ubuntu 20.04 LTS running ePC and remotely controlled eNB via PUTTY
1x USRP B210 connect to raspberry pi
1x COTS UE (Samsung Galaxy Neo 3)

## Expected Behavior ##
eNB successfully connect to ePC

## Actual Behaviour ##
eNB is not connected to ePC. 
SCTP is working normally.

`sctp                192.168.79.181:36412                               LISTEN`

## Additional Information ##

#####################################################################
                   srsENB configuration file
#####################################################################

[enb]
enb_id = 0x19B
cell_id = 0x01
phy_cell_id = 1
tac = 0x0007
mcc = 208
mnc = 92
mme_addr = 192.168.79.181
gtp_bind_addr = 192.168.79.133
s1c_bind_addr = 192.168.79.133
n_prb = 50
#tm = 1
#nof_ports = 1

[enb_files]
sib_config = sib.conf
rr_config  = rr.conf
drb_config = drb.conf

[rf]
dl_earfcn = 3625
tx_gain = 80
rx_gain = 40

device_name = UHD

device_args = auto

#####################################################################
                   srsEPC configuration file
#####################################################################

[mme]
mme_code = 0x1a
mme_group = 0x0001
tac = 0x0007
mcc = 208
mnc = 92
mme_bind_addr = 192.168.79.181
apn = test4g
dns_addr = 8.8.8.8
encryption_algo = EEA0
integrity_algo = EIA1
paging_timer = 2
request_imeisv = false
lac = 0x0006

[hss]
db_file = user_db.csv

[spgw]
gtpu_bind_addr   = 192.168.79.181
sgi_if_addr      = 172.16.0.1
sgi_if_name      = srs_spgw_sgi
max_paging_queue = 100

I hope someone can give me a solution to this problem.

Thanks.

kaposnick, 2023-01-09:
I have run into similar problems. Network traces (using wireshark) usually revealed the misconfiguration. It would be really helpful if you uploaded them.

sj5298, 2023-02-12:
Thank you. Issue solved.

closed 2023-02-12 by sj5298

athiravinod5gtechie, 2023-12-20:
Failed to bind SCTP socket to 172.22.0.37:0. Success
srsgnb_zmq  | srsGNB ERROR: Failed to create SCTP gateway.

I am getting this issue, while doing srsran setuo for 5g deployment using open5gs
how to solve this?

NagaJayaPraveen, 2024-02-02:
HI, this is praveen.

I have been trying to Download and build srsRAN 4G in ubuntu server, rasberry pi 4 device , but getting failure errors. please find the attached code

git clone https://github.com/srsRAN/srsRAN_4G.git
cd srsRAN_4G
mkdir build
cd build
cmake ../
make
make test

Here is error message:
[image]

Message:
 error: ‘void* __builtin_memmove(void*, const void*, long unsigned int)’ forming offset 36 is out of the bounds [0, 36] [-Werror=array-bounds=]
  437 |             __builtin_memmove(__result, __first, sizeof(_Tp) * _Num);
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors

Thanks,
Praveen

eirini5th, 2024-03-09:
`The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in dual executor mode.

--== srsRAN gNB (commit bcf941b34) ==--

Connecting to AMF on 127.0.1.100:38412
Failed to connect SCTP socket to 127.0.1.100:38412. error="Connection refused" timeout=0ms
srsGNB ERROR: Failed to create SCTP gateway.
`

I'm just trying to follow [the Running srsRAN guide from the project documentation](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/running.html) but I get this error when I run `sudo gnb -c gnb_rf_b200_tdd_n78_20mhz.yml`