srsran/srsRAN_4G #1081: EnvironmentError: IOError: x300 fw peek32 - reply timed out

RogerLin88124, 2023-01-10:
make error while using "uhd_usrp_probe"
but it can work on "uhd_find_device"
## Issue Description ##
[image]

[image]
## Setup Details ##
lab000@ubuntu:~/Desktop$ uhd_usrp_probe
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.3.0.0-0ubuntu1~focal1
[INFO] [X300] X300 initialization sequence...
[ERROR] [X300] 192.168.10.2: x300 fw communication failure #1
EnvironmentError: IOError: x300 fw peek32 - reply timed out
[ERROR] [X300] 192.168.10.2: x300 fw communication failure #2
EnvironmentError: IOError: x300 fw peek32 - reply timed out
[ERROR] [X300] 192.168.10.2: x300 fw communication failure #3
EnvironmentError: IOError: x300 fw peek32 - reply timed out
[ERROR] [X300] 192.168.10.2: x300 fw communication failure #1
EnvironmentError: IOError: x300 fw peek32 - reply timed out
[ERROR] [X300] 192.168.10.2: x300 fw communication failure #2
EnvironmentError: IOError: x300 fw peek32 - reply timed out
[ERROR] [X300] 192.168.10.2: x300 fw communication failure #3
EnvironmentError: IOError: x300 fw peek32 - reply timed out
Error: EnvironmentError: IOError: 192.168.10.2: x300 fw communication failure #3
EnvironmentError: IOError: x300 fw peek32 - reply timed out

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

HaiderHaideri, 2024-05-02:
@RogerLin88124 I am also facing this issue, what steps you have taken to solve this issue, will be waiting for your response soon

kazim425, 2024-09-30:
Upgrade to UHD 4.7. It will be fixed

Harsihv, 2024-10-03:
Just Interchange the Cables , thats enough for this issue

zhaojh074, 2024-10-19:
> @RogerLin88124 I am also facing this issue, what steps you have taken to solve this issue, will be waiting for your response soon

Did you solve this problem, I am facing the same issue and looking forward to your response

zhaojh074, 2024-10-19:
@Harsihv Did you solve this problem, I am facing the same issue and looking forward to your response @HaiderHaideri

zhaojh074, 2024-10-19:
Did you solve this problem, I am facing the same issue and looking forward to your response @HaiderHaideri

CNUzzx, 2025-06-12:
I am also facing this issue and looking forward to your response ,thanks! @HaiderHaideri