srsran/srsRAN_4G #1084: srsRAN snap created!

saltiyazan, 2023-01-18:
Hello team,

My name is Yazan Salti from the Telco team at Canonical.
We have created a snap which you can find [here](https://github.com/canonical/srsran-snap), to deploy srsRAN.
The snap should make the deployment of srsRAN easier as it will give you srsRAN and its dependencies as a bundle. Installing the snap on an ubuntu20.04 without any modification should be enough to have srsepc, srsenb and srsue working with the default configuration, which can be changed if needed. 
Soon the snap will be released on snap store.
[Here](https://snapcraft.io/docs) you can read more about snaps, snapcraft and snap store 

We'd like to contribute the snap to the project and give it to the community
Please feel free to get back to us with any feedback, comment or suggestion.

Thank you,

andrepuschmann, 2023-01-18:
Hey @saltiyazan, thanks for the heads-up. I noticed you're using srsRAN 20.10.1 for the snap. Any reason why you're not using the most recent version?

saltiyazan, 2023-01-20:
Hi @andrepuschmann, sorry for the late response, but I have been testing the most recent version `22.10`, and it works, but it also needs to build on ubuntu20.
I created a [PR](https://github.com/canonical/srsran-snap/pull/4) to address this change :)

closed 2023-04-03 by saltiyazan