srsran/srsRAN_4G #1088: Tx while waiting for EOB issue

StormOfL337, 2023-01-31:
## Issue Description ##
I have a X300 and a e320. Each one produces similar behavior and won't connect to each other (enb+ue) or (enb) to a pixel 6a cots ue.

I get this TX while waiting for EOB warning over and over on both my e320 and X300. On the e320, the RX2 LED flashes constantly and never turns solid. On the X300 the TX/RX LED flashes a lot and then becomes solid; however whenever I try to connect the e320 (via srsue) or a pixel 6a, it starts flashing again rapidly and repeats the EOB warnings.

Any help would be appreciated, thanks!

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

Asus ROG Laptop (i9-8950HK, 2.5gbe, nvidia rtx 2080 mobile)
UHD 4.3.0-HEAD-0-g1f8fd345
Rocky Linux 8.7
Ettus E320 and Ettus X300

## Expected Behavior ##
srsenb and srsue to connect

## Actual Behaviour ##
I get this behavior on my X300 as well:

[INFO] [UHD] linux; GNU C++ version 8.5.0 20210514 (Red Hat 8.5.0-16); Boost_106600; UHD_4.3.0.HEAD-0-g1f8fd345
[INFO] [LOGGING] Fastpath logging disabled at runtime.
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.10.3,type=e3xx,product=e320,serial=325EA60,name=ni-e320-325EA60,fpga=1G,claimed=False,addr=192.168.10.3
[INFO] [MPM.PeriphManager] init() called with device args `fpga=1G,mgmt_addr=192.168.10.3,name=ni-e320-325EA60,product=e320'.
[INFO] [0/Radio#0] Performing CODEC loopback test on channel 0 ... 
[INFO] [0/Radio#0] CODEC loopback test passed
[INFO] [0/Radio#0] Performing CODEC loopback test on channel 1 ... 
[INFO] [0/Radio#0] CODEC loopback test passed
[INFO] [0/DmaFIFO#0] BIST passed (Estimated Minimum Throughput: 1361 MB/s)
[INFO] [0/DmaFIFO#0] BIST passed (Estimated Minimum Throughput: 1361 MB/s)
[WARNING] [MULTI_USRP] Could not set RX rate to 11.520 MHz. Actual rate is 16.000 MHz
[WARNING] [MULTI_USRP] Could not set RX rate to 11.520 MHz. Actual rate is 16.000 MHz
[WARNING] [MULTI_USRP] Could not set TX rate to 11.520 MHz. Actual rate is 16.000 MHz
[WARNING] [MULTI_USRP] Could not set TX rate to 11.520 MHz. Actual rate is 16.000 MHz
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.696468 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 2.83964 >= 2.82837. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 4.85771 >= 4.85487. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 6.94006 >= 6.92813. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 8.95937 >= 8.95532. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 11.0552 >= 11.0295. Starting new burst...

## Steps to reproduce the problem ##
I run srsenb or srsue.

## Additional Information ##
As for the configuration parameters:

X300: type=x300,clock=internal,sampling_rate=11.52e6,send_frame_size=1472,recv_frame_size=1472,num_send_frames=64,num_recv_frames=64,addr=192.168.10.2

e320:
type=usrp2,sampling_rate=11.52e6,lo_freq_offset_hz=11.52e6,addr=192.168.10.3

wangde-fu, 2023-02-20:
Maybe you need to make sure that you connect your usrp device to the USB3.0 interface instead of 2.0. 
I once got a similar error like this：
```shell
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.xxx>= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 15.xxx >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 7.xxx >= 0. Starting new burst...
```
I solved this problem by reinserting my B210 into the USB3.0 interface

StormOfL337, 2023-02-27:
> Maybe you need to make sure that you connect your usrp device to the USB3.0 interface instead of 2.0. I once got a similar error like this：
>
> ```shell
> [INFO] [UHD RF] Tx while waiting for EOB, timed [...]

I think the e320 is usb 2.0 and ethernet/sfp+ only. The x300 is also running over SFP+ 10 gb and I'm not sure there is a usb for it.

But I could be wrong.

badf00d51, 2023-03-07:
Hii StormOfL337, could you resolve this problem. I am facing a similar issue with my E310.

closed 2023-06-01 by ismagom

kazim425, 2024-07-21:
> e320

10 Gb Streaming via SFP+ Port

Load the XG FPGA image for 10 Gb streaming as detailed in the section [Updating the FPGA Image](https://kb.ettus.com/E320_Getting_Started_Guide#Updating_the_FPGA_Image). You will need to use a 10 GigE cable that can be plugged in directly to the SFP+ connector on the board.

NOTE: The XG FPGA image must be loaded for the SFP+ Port to operate at 10 Gb speeds. If the 1G image is loaded, the port will be unresponsive at 10 Gb speeds. Mismatched MTU values on either the Host or E320 may cause flow control errors.