srsran/srsRAN_4G #1091: Does srsRAN support Network slicing?

jobgame, 2023-02-02:
## Issue Description ##
So I am starting to wonder does srsRAN support network slicing. Because I am working with open5gs, zmq,  and srsran. I got the basics working. However, now when I put a sd variable in the nssf and amf, open5gs just starts saying "imsi#  DNN not supported or Not subscribed in the slice" So I looked up that error and it says "This 5GMM cause is sent by the network to indicate that the requested service cannot be provided due to payload could not be forwarded by AMF because the DNN is not supported or not subscribed in the slice selected by the network **if the UE did not indicate a slice, or the DNN is not supported or not subscribed in the slice indicated by the UE.**".  So I looked at the ue.conf and there seems to be no network slicing variables available, which leads to my question.

## Setup Details ##
srsRAN, zmq, open5gs

## Expected Behavior ##
PDU session should get set in the srsue

## Actual Behaviour ##
It goes all the way to the message "Received RRC Release"  in the srsue and then Open5gs says  "imsi#  DNN not supported or Not subscribed in the slice" 

## Steps to reproduce the problem ##
Follow the srsRan example, but instead configure the sample.yaml in open5gs/build/configs then in the amf and nssf put the sd variable under sst like so:
sst: 1
 sd: 000001

Then create a script and put the following, then run it:

#!/bin/bash
sudo ip tuntap add name ogstun mode tun
sudo ip addr add 10.45.0.1/16 dev ogstun
sudo ip addr add 2001:db8:cafe::1/48 dev ogstun
sudo ip link set ogstun up
./build/tests/app/5gc

This will just run the sample.yaml and do that other stuff

mrouili, 2023-02-19:
Bumping this issue as we are facing the exact same problem, only difference is that we are using a USRP x310 as RF and google pixel 7 pro as UE.

VLiu7, 2023-02-21:
I have a similar problem, maybe the developers can answer it? @andrepuschmann 

Thanks a lot!

andrepuschmann, 2023-02-21:
Hey, there will be some updates to slicing coming with the next release. However, I would also suggest to keep an eye on the new 5G-only project called srsRAN Project which will be released very soon. At least those only working with the gNB are advised to use this repo instead.

andrepuschmann, 2023-02-21:
> @andrepuschmann Thanks for clarifying. But until the next release comes, is there any workaround for this with what's currently available? Assuming we are not trying to have RAN slicing but simply establish a successful connection. We are unable to do that because of this issue and we followed the COTE UE 5G SA guide.
> 
> thanks

The original issue is with srsUE. And we probably don't support the selection there properly. But you say you use a Pixel 7. So it must be something else, maybe in the core?