srsran/srsRAN_4G #1094: `srsue` attachs to cell successfully, but `sudo srsue` fails

wesmwoo, 2023-02-07:
## Issue Description ##
`sudo srsue` fails to attach to network, but `srsue` succeeds? 

## Setup Details ##
USRP B200 RF front end
magma as open source EPC
Baicells Nova227 TDD as eNB

## Expected Behavior ##
When I run `srsue`, I receive the error
```
Network attach successful. IP: 192.168.128.22
Failed to setup/configure GW interface
```
To solve this, I want to run `sudo srsue`, which will give srsue the permission to create the TUN. 

## Actual Behaviour ##
I cannot find the cell when I run `sudo srsue`:
```
Waiting PHY to initialize ... done!
Attaching UE...
...................
.....
...................
....
.
.......
...
.......
```
until I close the program.

## Steps to reproduce the problem ##
Used the basic quick-start setup of magma as described [here](https://magma.github.io/magma/docs/basics/quick_start_guide.html) and [here](https://magma.github.io/magma/docs/next/lte/dev_notes#connecting-a-physical-enodeb-and-ue-to-gateway-vm).

ue.conf file:
[ue.conf.txt](https://github.com/srsran/srsRAN/files/10676377/ue.conf.txt)

ninjab3s, 2023-03-24:
Can you provide the full console output of both commands, including the command itself?

ehitimum, 2023-11-19:
For srsue: root@srsRAN:~/.config/srsran# sudo srsue
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Couldn't open , trying /root/.config/srsran/ue.conf
Reading configuration file /root/.config/srsran/ue.conf...

Built in Release mode using commit fa56836b1 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Waiting PHY to initialize ... done!
Attaching UE...
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.2 KHz
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Found PLMN:  Id=31501, TAC=1
Random Access Transmission: seq=41, tti=1621, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=0
Network attach successful. IP: 10.45.0.2
Failed to setup/configure GW interface
Killed
root@srsRAN:~/.config/srsran# ^C
root@srsRAN:~/.config/srsran# nano ue.conf 
root@srsRAN:~/.config/srsran# sudo srsue
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Couldn't open , trying /root/.config/srsran/ue.conf
Reading configuration file /root/.config/srsran/ue.conf...

Built in Release mode using commit fa56836b1 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Waiting PHY to initialize ... done!
Attaching UE...

For srsenb:
root@srsRAN:~/.config/srsran# srsenb
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /root/.config/srsran/enb.conf
Reading configuration file /root/.config/srsran/enb.conf...

Built in Release mode using commit fa56836b1 on branch master.

Opening 1 channels in RF device=zmq with args=fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=enb
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2000
CH0 fail_on_disconnect=true

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
RACH:  tti=1621, cc=0, pci=1, preamble=41, offset=0, temp_crnti=0x46
Killed
root@srsRAN:~/.config/srsran# 

And mme.log of my open5gs EPC: 
11/19 08:07:53.712: [mme] INFO: [Added] Number of eNBs is now 1 (../src/mme/mme-context.c:2557)
11/19 08:07:53.712: [mme] INFO: eNB-S1[192.168.0.97] max_num_of_ostreams : 30 (../src/mme/mme-sm.c:150)
11/19 08:08:11.947: [mme] INFO: InitialUEMessage (../src/mme/s1ap-handler.c:402)
11/19 08:08:11.947: [mme] INFO: [Added] Number of eNB-UEs is now 1 (../src/mme/mme-context.c:4433)
11/19 08:08:11.947: [mme] INFO:     ENB_UE_S1AP_ID[1] MME_UE_S1AP_ID[1] TAC[1] CellID[0x19b01] (../src/mme/s1ap-handler.c:578)
11/19 08:08:11.947: [mme] INFO: [315010123456065] Unknown UE by IMSI (../src/mme/mme-context.c:3268)
11/19 08:08:11.947: [mme] INFO: [Added] Number of MME-UEs is now 1 (../src/mme/mme-context.c:3089)
11/19 08:08:11.947: [emm] INFO: [] Attach request (../src/mme/emm-sm.c:412)
11/19 08:08:11.947: [emm] INFO:     IMSI[315010123456065] (../src/mme/emm-handler.c:234)
11/19 08:08:12.024: [mme] INFO: [Added] Number of MME-Sessions is now 1 (../src/mme/mme-context.c:4447)
11/19 08:08:12.363: [mme] INFO: eNB-S1[192.168.0.97] connection refused!!! (../src/mme/mme-sm.c:166)
11/19 08:08:12.363: [mme] INFO: [Removed] Number of eNB-UEs is now 0 (../src/mme/mme-context.c:4440)
11/19 08:08:12.364: [mme] INFO: [Removed] Number of eNBs is now 0 (../src/mme/mme-context.c:2591)
11/19 08:08:18.077: [mme] ERROR: S1 context has already been removed (../src/mme/nas-path.c:90)
11/19 08:08:18.077: [emm] ERROR: emm_state_initial_context_setup: Expectation `r == OGS_OK' failed. (../src/mme/emm-sm.c:1491)
11/19 08:08:24.079: [mme] ERROR: S1 context has already been removed (../src/mme/nas-path.c:90)
11/19 08:08:24.080: [emm] ERROR: emm_state_initial_context_setup: Expectation `r == OGS_OK' failed. (../src/mme/emm-sm.c:1491)
11/19 08:08:30.086: [mme] ERROR: S1 context has already been removed (../src/mme/nas-path.c:90)
11/19 08:08:30.086: [emm] ERROR: emm_state_initial_context_setup: Expectation `r == OGS_OK' failed. (../src/mme/emm-sm.c:1491)
11/19 08:08:36.088: [mme] ERROR: S1 context has already been removed (../src/mme/nas-path.c:90)
11/19 08:08:36.089: [emm] ERROR: emm_state_initial_context_setup: Expectation `r == OGS_OK' failed. (../src/mme/emm-sm.c:1491)
11/19 08:08:42.097: [emm] WARNING: Retransmission of IMSI[315010123456065] failed. Stop retransmission (../src/mme/emm-sm.c:1466)
^C

My enb do managed to connect with the epc successfully but my ue can't connect properly. The same error is given and if I try again it will stuck on the "Attaching UE..." part.