srsran/srsRAN_4G #1096: COTS UE cannot see the LTE network

sj5298, 2023-02-12:
## Issue Description ##
Hi all,

I run sudo srsepc and sudo srsenb, the eNB and ePC are connected but the corresponding network cannot be found using COTS UE (Samsung Galaxy Neo 3). I am aware that the phone I used supports GSM900, so I have used the DL_EARFCN in the range from 3450 to 3790. I have changed the value of DL_EARFCN but it does not work.

## Setup Details ##
My setup was listed as follows:
1x Raspberry pi as eNB
1x laptop with ubuntu 20.04 LTS running ePC and remotely controlled eNB via PUTTY
1x USRP B210 connect to raspberry pi
1x COTS UE (Samsung Galaxy Neo 3)

## Expected Behavior ##
COTS UE could find the LTE network at 925MHz and connect it.

## Actual Behaviour ##
COTS UE couldn't find the network.

[image]
[image]

## Additional Information ##

[enb.txt](https://github.com/srsran/srsRAN/files/10715675/enb.txt)
[epc.txt](https://github.com/srsran/srsRAN/files/10715677/epc.txt)
[user_db.txt](https://github.com/srsran/srsRAN/files/10715681/user_db.txt)
[USIM card details.txt](https://github.com/srsran/srsRAN/files/10715682/USIM.card.details.txt)

I hope someone can give me a solution to this problem.

Thanks.

zikkuratti, 2023-02-15:
kernel version plz?

sj5298, 2023-02-16:
> kernel version plz?

Hi zikkuratti,

The kernel version of laptop (EPC):
Linux seah-X450LD 5.15.0-60-generic #66-Ubuntu SMP Fri Jan 20 14:29:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

The kernel version of Raspberry Pi(eNB):
Linux ubuntu 5.4.0-1080-raspi #91-Ubuntu SMP PREEMPT Thu Jan 19 09:35:03 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

bedrankara, 2023-02-16:
After quickly checking the Samsung Galaxy Neo 3 specs, it seems like it does not support LTE ?

sj5298, 2023-02-17:
> After quickly checking the Samsung Galaxy Neo 3 specs, it seems like it does not support LTE ?

It supported 4G based on [Samsung website](https://www.samsung.com/ie/smartphones/others/galaxy-s3-i9305-gt-i9305rwr3ie/)

I did also tried with Samsung note 10 lite. Still couldn't see the LTE network.

mrahman2-vt, 2023-02-17:
Hi there,
I would like to implement a scenario with CU DU split mode in Raspberry Pi,
and my propagation delay between CU and DU is quite large, say for example
6ms. I think the RACH procedure will fail because of this huge delay. Is
there any way I can modify the RACH procedure? If so, can you please guide
me on exactly what I need to make those changes?

*Regards*

*MdMahfuzur Rahman (Mahfuz)*

*+1-906-281-8261*

***@***.*** ***@***.***>*

Message ID: ***@***.***>
>

zikkuratti, 2023-02-17:
@sj5298  saw some troubles on PREEMT try to patch on PREEMT_RT dont ask how) work on same problem for arm64 device

dufffffy, 2023-04-10:
Did you solve the problem? I met the same with you.

NagaJayaPraveen, 2024-02-02:
HI, this is praveen.

I have been trying to Download and build srsRAN 4G in ubuntu server, rasberry pi 4 device , but getting failure errors. please find the attached code 

git clone https://github.com/srsRAN/srsRAN_4G.git
cd srsRAN_4G
mkdir build
cd build
cmake ../
make
make test
------------------- 
Here is the error message:

[image]

error message:

 error: ‘void* __builtin_memmove(void*, const void*, long unsigned int)’ forming offset 36 is out of the bounds [0, 36] [-Werror=array-bounds=]
  437 |             __builtin_memmove(__result, __first, sizeof(_Tp) * _Num);
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors

could you help me on this!

Thanks,
Praveen