srsran/srsRAN_4G #1099: Warning: Failed to create thread with real-time priority. Creating it with normal priority: Resource temporarily unavailable

egortaran, 2023-02-15:
## Issue Description
Hello dear community. I have a problem, after writing commands in the terminal I have this log:

<details>
  <summary>$ sudo srsenb --expert.lte_sample_rates=true /etc/srsran/enb.conf </summary>

  ```
Starting srsenb ... done
Opening 2 channels in RF device=soapy with args=rxant=LNAH,txant=BAND1
Soapy has found device #0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 908340189241A, media=USB 3.0, module=FX3, name=LimeSDR-USB, serial=000908340189241A, 
Selecting Soapy device: 0
Detected LimeSDR. Consider using LTE rates for better RF performance.
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 908340189241A'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M register cache: Disabled
Setting up Rx stream with 2 channel(s)
Setting up Tx stream with 2 channel(s)
[INFO] RX LPF configured
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
Available sensors for Rx channel 1: 
 - lo_locked
Setting Rx channel 0 antenna to LNAH
Setting Rx channel 1 antenna to LNAH
Setting Tx channel 0 antenna to BAND1
Setting Tx channel 1 antenna to BAND1
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -7.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 0.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAH
Tx antenna set to BAND1
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Resource temporarily unavailable
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Resource temporarily unavailable
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Resource temporarily unavailable
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Resource temporarily unavailable
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Resource temporarily unavailable

==== eNodeB started ===
Type <t> to view trace
[INFO] RX LPF configured
[INFO] RX LPF configured
Setting frequency: DL=942.5 Mhz, UL=897.5 MHz for cc_idx=0 nof_prb=15
[INFO] Tx calibration finished
[INFO] Tx calibration finished
[INFO] Rx calibration finished
[INFO] Rx calibration finished

  ```
</details>
The cell phone does not catch the network. With the help of the analyzer, I found that LimeSDR does not emit a signal. I guess the problem is 

**Warning: Failed to create thread with real-time priority. Creating it with normal priority: Resource temporarily unavailable** 

What should be research or changed to fix this error?

* * *
[... 842 lines cut ...]
    carrier_freqs_info_list =
    (
        {
            cell_resel_prio = 0;
            ncc_permitted = 255;
            q_rx_lev_min = 0;
            thresh_x_high = 2;
            thresh_x_low = 2;

            start_arfcn = 871;
            band_ind = "dcs1800";
            explicit_list_of_arfcns = (
                871
            );
        }
    );
};
  
  ```
</details>

## Technology stack
Ubuntu 22.04 
srsRAN 22.04 (open5gs)
LimeSDR v1.4s (LimeSuite v20.10.0, SoapySDR 0.7.2)

egortaran, 2023-02-27:
@andrepuschmann, sorry to disturb you. The problem remains unresolved, I will be very grateful for the answer

andrepuschmann, 2023-03-03:
Usually the issue comes from insufficient rights to set scheduler to RT. Since you're running with sudo you should have the rights though. I don't think it's an srsRAN issue though.

wangde-fu, 2024-04-05:
I can start srsenb when I use normal user rights, but I will get the same error as you. When I start srsenb with the root user, it prompt "error while loading shared libraries: libsrsran_rf.so.0". 
I encountered the same problem and solved it in the following ways. Maybe you can try.
1. Run on the command line terminal:“find / -name libsrsran_rf.so.0”. I found it under my project folder and "/ usr/local/lib" folder.
2. Run on the command line terminal:"vi /etc/ld.so.conf", and add "/ usr/local/lib" under "include ld.so.conf.d/*.conf".
3. After save it, run on the command line terminal:"sudo /sbin/ldconfig -v", "sudo ldconfig"
It works for me.
---------------
Ubuntu 22.04
srsRAN 23.04 
usrp b210

zahrayh, 2025-01-11:
did you solve this problem ? I have the same issue.