srsran/srsRAN_4G #1101: Does not work with Bladerf 2.0 micro

pricalinat, 2023-02-18:
## Issue Description ##
I tried to use Bladerf 2.0 micro to connect, but found that it was stuck when running.
## Setup Details ##
[image]

[image]

[image]

[image]

use default conf (sample)

[image]
[image]
(And my phone doesn't have any signal, I don't know if it has something to do with it)

srsRAN version:latest

## Expected Behavior ##
I want it to work like zmq...

## Actual Behaviour ##
stuck for a long time

pricalinat, 2023-02-18:
I follow this https://docs.srsran.com/en/latest/app_notes/source/cots_ue/source/index.html#introduction
when doing "Under the Network Operators find the network which you have just instantiated using srsRAN"
I can not find my srsran network

closed 2023-02-19 by pricalinat

gustavobsch, 2023-03-11:
hi @pricalinat , did you resolve this issue? also, may I ask about the  SIM card write program? where did you get the new 4.4.3 version from? thanks!