srsran/srsRAN_4G #1102: COTS UE can connect to the network, but cannot access the Internet

pricalinat, 2023-02-19:
## Issue Description ##
UE can connect to the network, but cannot access the Internet.
The signal displayed by the UE is full, but the Internet cannot be accessed.
## Setup Details ##
srsRAN 22.10
Ubuntu 20.04
5G programmable sim card
4G xiaomi Mi 8 
bladerf 2.0 micro

ENB
[image]
EPC
[image]
[image]

UL NAS: PDN Connectivity Request
DL NAS: Sending PDN Connectivity Reject

LOG
[log&pcap.zip](https://github.com/srsran/srsRAN/files/10775671/log.pcap.zip)

## Expected Behavior ##
access the Inteenet

ninjab3s, 2023-03-24:
Hello pricalinat,
to me it seems like you are missing the APN on your phone. The APN has to match the one you set in your epc.conf, default is `srsapn`. Furthermore, you also need to forward traffic from your local interface to the internet and vice versa. To do so you can use the script `srsepc_if_masq.sh`.
If this doesn't work, please provide logs from epc and enb of the attach on debugging level and in text form please.

AlvaSoto, 2023-06-11:
Hello, I have the same problem. I have secured the APN and run the masking script with the interface that gives access to the internet. The mobile connects to the network but does not have internet access.

Did you manage to solve the problem?

c0nfused-penguin, 2023-07-06:
Hi, I have the same problem. I can connect to the network on my iPhone but no access to internet. I have set the APN on my phone to the one in the epc.conf file.

```
root@enbepc-HP-EliteBook-840-G7-Notebook-PC:/home/enb-epc/Desktop# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         router.asus.com 0.0.0.0         UG    600    0        0 wlp0s20f3
link-local      0.0.0.0         255.255.0.0     U     1000   0        0 wlp0s20f3
192.168.72.0    0.0.0.0         255.255.255.0   U     600    0        0 wlp0s20f3
root@enbepc-HP-EliteBook-840-G7-Notebook-PC:/home/enb-epc/Desktop# srsepc

Built in RelWithDebInfo mode using 22.04.0.

---  Software Radio Systems EPC  ---

Couldn't open , trying /root/.config/srsran/epc.conf
Reading configuration file /root/.config/srsran/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf901, MNC: 0xff70
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:901, MNC:70
S1 Setup Request - TAC 7, B-PLMN 0x9f107
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0xd31598bb
Attach request -- eNB-UE S1AP Id: 1
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 43
PDN Connectivity Request -- ESM Information Transfer requested: false
UL NAS: Received Identity Response
ID Response -- IMSI: 001010123456789
Downlink NAS: Sent Authentication Request
UL NAS: Received Authentication Response
Authentication Response -- IMSI 001010123456789
UE Authentication Accepted.
Generating KeNB with UL NAS COUNT: 0
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 001010123456789
Getting subscription information -- QCI 9
Sending Create Session Request.
Creating Session Response -- IMSI: 1010123456789
Creating Session Response -- MME control TEID: 1
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 1
SPGW: Allocated User TEID 1
SPGW: Allocate UE IP 172.16.0.2
Received Create Session Response
Create Session Response -- SPGW control TEID 1
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 172.16.0.2 to IMSI 001010123456789
Adding attach accept to Initial Context Setup Request
Sent Initial Context Setup Request. E-RAB id 5 
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x1; eNB GTP-U Address 127.0.1.1
UL NAS: Received Attach Complete
Unpacked Attached Complete Message. IMSI 1010123456789
Unpacked Activate Default EPS Bearer message. EPS Bearer id 5
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Sending EMM Information
Received UE Context Release Request. MME-UE S1AP Id 1
There are active E-RABs, send release access bearers request
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
Received UE Context Release Complete. MME-UE S1AP Id 1
UE Context Release Completed.
```
```
root@enbepc-HP-EliteBook-840-G7-Notebook-PC:/home/enb-epc/Desktop# srsepc_if_masq.sh wlp0s20f3
Masquerading Interface wlp0s20f3
root@enbepc-HP-EliteBook-840-G7-Notebook-PC:/home/enb-epc/Desktop# srsenb
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /root/.config/srsran/enb.conf
Reading configuration file /root/.config/srsran/enb.conf...
Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
Couldn't open rr.conf, trying /root/.config/srsran/rr.conf
Couldn't open rb.conf, trying /root/.config/srsran/rb.conf
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in RelWithDebInfo mode using 22.04.0.

/build/srsran-j7rCyf/srsran-22.04/srsenb/src/enb_cfg_parser.cc:1507: Force DL EARFCN for cell PCI=1 to 8865
Opening 1 channels in RF device=default with args=default
Supported RF device list: UHD file
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_3.15.0.0-74-ge35f66e8
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=876.5 Mhz, UL=831.5 MHz for cc_idx=0 nof_prb=50
RACH:  tti=2961, cc=0, pci=1, preamble=42, offset=1, temp_crnti=0x46
User 0x46 connected
Disconnecting rnti=0x46.
```
Any ideas what the problem could be? Thanks!

BouazzaEhm, 2024-03-28:
any news ? 
I have the same problem

ninjab3s, 2024-03-30:
Hello @BouazzaEhm , @c0nfused-penguin , @AlvaSoto,

please record a PCAP of the EPC and share it here. I also need the EPC log and config. Thanks.

BouazzaEhm, 2024-03-30:
Hello sir i really thank you for helping me because u truly need this work
here is the log and config files  
[epc.log](https://github.com/srsran/srsRAN_4G/files/14812542/epc.log)
[epcconfig.log](https://github.com/srsran/srsRAN_4G/files/14812556/epcconfig.log)

BouazzaEhm, 2024-03-30:
[image]
[image]
[image]

BouazzaEhm, 2024-03-30:
[image]

BouazzaEhm, 2024-04-01:
@ninjab3s  any news sir 
i am very sorry but its for my masters degree and i truly need this

ninjab3s, 2024-04-01:
@BouazzaEhm thanks for the effort. Can you please upload the pcap files here? I cannot really check whats going on based on the screenshots. Thanks!

BouazzaEhm, 2024-04-02:
@ninjab3s

BouazzaEhm, 2024-04-02:
Hello @ninjab3s here os the PCAP file ill hope youll help me with this 
[EPC.zip](https://github.com/srsran/srsRAN_4G/files/14837845/EPC.zip)

BouazzaEhm, 2024-04-02:
Someone told me that i should have an interface for the EPC in the ifconfig command is that true 
because i didnt find any of this in the installation guide

ninjab3s, 2024-04-03:
Hi @BouazzaEhm.
Looking at the EPC log I see the following line:
`2024-03-30T15:48:43.495686 [NAS    ] [I] ESM Info: APN oai.ipv4`

Are you sure that the APN in the UE is activated and set to srsapn, like you did in your EPC conf?

In the PCAP you provided I see a "PDU connectivity reject". This is probably caused by the wrong APN.

Please make sure that the APN in the phone is configured and **enabled**. Do you have another UE you could use to test?

BouazzaEhm, 2024-04-04:
i think that is the problem 
i did not get any interface when installing srsEPC and i intalled it many times