srsran/srsRAN_4G #1104: 5G SA COTS UE cannot see the network

pricalinat, 2023-02-20:
https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html
I built open5gs+srsran according to the steps of this website, trying to build a 5G network. But when I use UE to connect, UE cannot find the network I built.
I am using bladeRF 2.0 micro xA4, has anyone tried to connect to 5G network successfully with this device? I don't know if it's an issue with the rf device or the phone since my phone isn't rooted yet.

JoseSecmotic, 2023-02-24:
Were you successful connecting first to a 4G network? I would start trying that implementation first, so then you can narrow down better the issues

pricalinat, 2023-02-24:
> Were you successful connecting first to a 4G network? I would start trying that implementation first, so then you can narrow down better the issues

Thanks for your reply, I have successfully done it in 4G network. 
There are three mobile devices, Apple directly succeeded, an Android phone succeeded after modifying earfcn, and a Huawei has not been successful.
But in 5G network, I tried but failed.
Here is my log in 5G network.
[image]
[enb.log](https://github.com/srsran/srsRAN_4G/files/10822484/enb.log)

JoseSecmotic, 2023-02-24:
What I would try to do first, is to put the CPU in performance mode. You can do it by the following command:

` echo "performance" | sudo tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`

Anyway, for me it seems to be an issue with the COTS UE. What SIMs are you using? I am stuck at the same point

pricalinat, 2023-02-24:
Thank you for your suggestion, but my CPU is already in performance mode. 
I'm also inclined to suspect that it's a COTS UE issue, but I haven't found a solution yet.
And you reminded me, I used an unbranded 5G test card. I'm not sure if that's the cause, I'll take a  look.

JoseSecmotic, 2023-02-24:
> Thank you for your suggestion, but my CPU is already in performance mode.
> I'm also inclined to suspect that it's a COTS UE issue, but I haven't found a solution yet.
> And you reminded me, I used an unbranded 5G test card. I'm not sure if that's the cause, I'll take a look.

Well, according to your logs, the CPU is not in performance mode (check the warning lines).
Regarding the USIMs, there might be needed to program them and deactivate some of the 5G fields in the SIM card. Are your SIMs from sysmocom? or is it another provider?

Check out [this guide](https://webcache.googleusercontent.com/search?q=cache:xmC6RcZZvDMJ:https://docs.srsran.com/en/latest/app_notes/source/cots_ue/source/index.html&cd=2&hl=es&ct=clnk&gl=es)

pricalinat, 2023-02-24:
The sim card I used was a cheap five dollar one made by a small private company. 
I can modify the following parameters through software.
[image]

iTzJP-Sec, 2023-04-13:
In these days I will deploy the same network architecture with the same equipment. @JoseSecmotic @pricalinat Maybe you can give me some guidance and let me know if you have already solved the problem.

tiger762, 2023-04-20:
Nothing to add other than I am stuck at the same point you are.

IPhone 12, iOS 16.4.1, dual core i5 Kali Linux, Ettus B200, external 10MHz GPSDO, srsENB works perfectly. When I run 'gnb' I do get some underflows if I let it run for a while. Any help appreciated! 

└─# gnb -c gnb_test_n2.yml
Available radio types: uhd.

--== srsRAN gNB (commit fbe73a49c) ==--

Connecting to AMF on 192.168.16.254:38412
[INFO] [UHD] linux; GNU C++ version 12.2.0; Boost_107400; UHD_4.3.0.0+ds1-4+b2
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200,num_recv_frames=64,num_send_frames=64,clock=external'
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Cell pci=1, bw=5 MHz, dl_arfcn=391500 (n2), dl_freq=1957.5 MHz, dl_ssb_arfcn=391470, ul_freq=1877.5 MHz

==== gNodeB started ===
Type <t> to view trace
Late: 2; Underflow: 35; Overflow: 0;
Late: 5; Underflow: 33; Overflow: 0;
Late: 14; Underflow: 9; Overflow: 0;
Late: 1; Underflow: 7; Overflow: 0;
Late: 2; Underflow: 22; Overflow: 0;
Late: 1; Underflow: 4; Overflow: 0;
Late: 0; Underflow: 2; Overflow: 0;
Late: 12; Underflow: 8; Overflow: 0;

ismagom, 2023-04-21:
Hi @tiger762 , 

First of all, please use https://github.com/srsran/srsRAN_Project/discussions to get support with srsRAN_Project. 

I think your problem is the sampling rate is too high. If you are using 5 MHz cell, set `srate` option in `rf_driver` section of the gnb.yml file to 5.76

If you still want to run 23.04 sampling rate, then you should set the following options (in the same section): 
```
device_args: type=b200,num_recv_frames=64,num_send_frames=64
otw_format: sc12
```

tiger762, 2023-04-21:
OK, I finally realized that I am in the wrong discussion group. Indeed, I have been using srsRAN_Project but confused the two groups.

My overall objective is to reduce the load on the CPU, hence the 5MHz B/W. Using a sampling rate of less than twice the bandwidth is what confuses me. I tried it, and got the "late/underflow/overflow" more often than when using 11.52 or 23.04, FWIW...

ismagom, 2023-04-22:
This is complex sampling in baseband, so 5.76 MHz will be good. It's weird you have lates. What device_args are you using?

tiger762, 2023-04-22:
Making USRP object with args 'type=b200,num_recv_frames=64,num_send_frames=64,clock=external'

I am guessing that this IPhone is indeed looking over a reduced range of frequencies, notwithstanding Apple's literature which shows coverage in all these 5G bands:

5G NR (Bands n1, n2, n3, n5, n7, n8, n12, n20, n25, n28, n38, n40, n41, n66, n71, n77, n78, n79)
5G NR mmWave (Bands n260, n261)

When it was originally sold to T-Mobile, they may have told Apple what specific frequencies to look for, to speed up the network discovery process. I don't know.

I have a OnePlus Nord 5G on order. Hopefully it will be less troublesome.

iTzJP-Sec, 2023-07-14:
Hi @ismagom, I have the same problem and I get: "late/underflow/overflow". have you managed to solve it? I use the example file gnb_rf_b200_tdd_n78_20mhz.yml without modifications.

iTzJP-Sec, 2023-07-14:
Hi @tiger762, I have the same problem and I get: "late/underflow/overflow". have you managed to solve it? I use the example file gnb_rf_b200_tdd_n78_20mhz.yml without modifications.

aldalahmeh, 2023-07-18:
The underflow/ overflow can be fixed if you run sudo gnb ....

iTzJP-Sec, 2023-07-18:
I run sudo and it doesn't work for me

houshangaziziaccelleran, 2024-04-19:
Have you managed to fix it?

Moussa-Guemdani, 2024-04-23:
Hello,
I have same problem, I am using USRP x310, n77 band, samsung s23.

[image]

Alan1321, 2025-04-02:
Any solutions? I have the same issue.