srsran/srsRAN_4G #1105: Found a way to use bladeRF native driver for srsUE. (It provides stable RRC connection)

DancingSW, 2023-02-22:
## Issue Description ##
I am writing this post to share my experience.

My bladeRF A4 have been working well when it used as srsENB/. However, I had a problem when using bladeRF A4 as srsUE. The issue was as follows:
- When I used bladeRF native driver, the srsUE stopped after it found the cell.
- When I used SoapySDR with bladeRF plugin, the RRC connection was hit or miss.

I solved this problem. Now, I am using the native bladeRF driver for srsUE and the connection is stable.
Followings are what I have done for solving my problem.

1. Changed the "srate " as 11.52e6 in **ue.conf**
2. Changed the  default "tx_srate" and "rx_srate" as 11.52.e6 in **rf_blade_imp.c**. You can find **rf_blade_imp.c** at _/srsRAN/lib/src/phy/rf/_. Then, I recompiled the srsRAN.

**(Example)**
```
225   /* Set default sampling rates */ 
226   rf_blade_set_tx_srate(handler, 11.52e6);
227   rf_blade_set_rx_srate(handler, 11.52e6);
```
It seems it works well if the default srate is same with the srate I want to use.
I spent almost a week to solve this problem and could not find any helpful post. Hope this can help someone who has same problem as me.

DancingSW, 2023-03-29:
> Hello ! I have the same issue as you had. I'm using 2 bladeRFs : one is used as srsUE and the other one as srsENB. I'm running the srsUE file and srsENB+srsEPC files ont two different PCs. I used your tip to try to handle my probem but it still persists : UE doesn't synchronize. Do you have any idea why doesn't it work? I attached my .conf and logs here if you can take a look :) Thank you for sharing your experience in any case!
> 
> P.S. I tried to install Soapy Driver but while using it my bladeRF device is not found. I also tried to change tx and rx gains and I switched cpu to performance mode.
> 
> If you have any ideas I will be very grateful!
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [ue2.log](https://github.com/srsran/srsRAN_4G/files/11087258/ue2.log)

Can you show your eNB config?
In addition, I remember that downlink earfcn of 3350 was not stable.  In my case, I used 2450 or 2500 in the lab environment where it does not affect the operator's networks. Please check the frequency policy in your country before you change it.

CatiaS11, 2023-03-30:
Thank you for your anwer, I will try to change my downlink earfcn !

Here is my eNB config file :

[image]

[image]

[image]

[image]

[image]

[image]

[image]

CatiaS11, 2023-03-30:
[enb.zip](https://github.com/srsran/srsRAN_4G/files/11109991/enb.zip)
[epc.zip](https://github.com/srsran/srsRAN_4G/files/11109992/epc.zip)
[ue.zip](https://github.com/srsran/srsRAN_4G/files/11109993/ue.zip)

DancingSW, 2023-03-30:
@CatiaS11
Please try to use a native bladeRF driver, not Soapy with bladeRF plugin. You may need to revise rf_blade_imp.c at /srsRAN/lib/src/phy/rf/ as I suggested in the first post.

CatiaS11, 2023-03-31:
@DancingSW Thanks you for your suggestion! Actually I have already tried to change rf_blade_imp.c file, but it didn't work well for me. May I ask you a stupid question (I'm a beginner) but how do you check which driver is used ?

DancingSW, 2023-03-31:
Did you recompile the srsRAN after you revise the code like _make, sudo make install_?
And, please share your logs and describe what you have done and what you have got after you try. 
I cannot help you without any clue.

You will directly check if you are using bladeRF driver when you running the srsUE or srsENB on the terminal....