srsran/srsRAN_4G #1107: 5G SA/NSA downlink "carrier" is not continuous or unstable

AndrewOG255, 2023-02-23:
## Issue Description ##
Hi All! I have been trying to bring up srsRAN in 5G SA COTS UE mode during the last month. I have been googling and studying all the possible github srsRAN issues here and lots of documentation. I made lots of optimizations like low latency kernel and CPU frequency scaling governor and downgraded UHD driver to 3.15 LTS and ... and ... and srsRAN still does not work correctly in 5G mode. But it works excellent when I bring up LTE 20Mhz on the same hardware.

I have some test equipment and Keysight VSA 89600 (Vector Signal Analyzer) clearly shows that srsRAN emits "carrier" (subcarriers) during very short time ... it emits like during SIB1 transmission only and then does not emits anything during the rest time like srsenb is stopped. Because of this typical COTS UE do not see the network. But we have COTS UE with special software/firmware that can lock to a specific channel and scan constantly only withing the specified channel.  So, these special UEs can "see" the network but no one can't connect it and register.

I have the same problem with both 5G SA and 5G NSA modes. But, I have no problem when I run LTE.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

HOST HARDWARE: BOXNUC7i5BNK
https://www.minipc.de/de/catalog/il/2174?gclid=Cj0KCQiAutyfBhCMARIsAMgcRJT5TGxIUyKU6xEMITHaHIx3gQhOwLHBhGjePkCK4dQzVBqGBqd403kaAgJrEALw_wcB

srsRAN: 22.10 release (manually compiled after UHD driver installed)
open5gs:  open5gs-2.6.0 release (manually compiled)

GPSDO: Trimble GPSDO from an industrial Samsung LTE Basestation

SDR: USRP B205mini-i
UHD: 3.15 LTS UHD driver (manually compiled) ... I also tested with the latest 4.4.0.0 UHD drivers (manually compiled and Ettus  compiled binaries) ... the problem's behavior is the same (no matter what UHD driver version is)

`root@srv02:~# uhd_usrp_probe
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_3.15.0.0-0-unknown
[INFO] [B200] Detected Device: B205mini
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz...
[INFO] [B200] Actually got clock rate 16.000000 MHz.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B205mini
|   |   serial: 327AA72
|   |   name: B205i
|   |   product: 30522
|   |   revision: 6
|   |   FW Version: 8.0
|   |   FPGA Version: 7.0
|   |
|   |   Time sources:  none, internal, external
|   |   Clock sources: internal, external
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX1
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
[... 142 lines cut ...]
Setting frequency: DL=2140.0 Mhz, DL_SSB=2139.85 Mhz (SSB-ARFCN=427970), UL=1950.0 MHz for cc_idx=0 nof_prb=52
t
Enter t to stop trace.
`

## Expected Behavior ##
Normal waterfall spectrum with continuous emission of sync signals

## Actual Behaviour ##
Randomly emits SIB1 and no continuous emission of sync signals

## Steps to reproduce the problem ##
No matter what 5g configuration I specify (SA or NSA). I always have the described above behavior.

## Additional Information ##
See the files attached
[image]
[image]

[rr.conf.txt](https://github.com/srsran/srsRAN_4G/files/10814909/rr.conf.txt)
[enb.conf.txt](https://github.com/srsran/srsRAN_4G/files/10814914/enb.conf.txt)
[amf.yaml.txt](https://github.com/srsran/srsRAN_4G/files/10814925/amf.yaml.txt)
[upf.yaml.txt](https://github.com/srsran/srsRAN_4G/files/10814957/upf.yaml.txt)
[smf.yaml.txt](https://github.com/srsran/srsRAN_4G/files/10814975/smf.yaml.txt)
[enb.log.zip](https://github.com/srsran/srsRAN_4G/files/10814991/enb.log.zip)

AndrewOG255, 2023-03-12:
UPDATE:
 - we have found that some COTS UEs passes RRCSetup and RRCSetupComplete and fails on authentication because we use COTS SIM cards that are not designed for such tests (some of them were purchased when 5G was not released) ... for example ASUS ROG 3 successfully passes but it has quite old modem firmware
 - Samsung Galaxy S23+ has the latest modem firmware updates and fails between RRCSetup and RRCSetupComplete
 - After RRCSetup UE starts lots of downlink channel estimations and validations
 - We enabled UE maximum logging details and this is what we found: **RRC-ML1 config validate: Validation failed on CSI-RS params**
```
NR5GML1   	nr5g_ml1_rrc_intf.c:            RRC-ML1 config validate: BWP cfg validation
 . . .
NR5GML1   	nr5g_ml1_rrc_intf.c:            RRC-ML1 config validate: BWP ID 0 Current CORESET mask 0x1 sspace_mask 0x0000000000000003
NR5GML1   	nr5g_ml1_rrc_intf.c:            RRC-ML1 config validate:After release for BWP ID 0, CORESET mask 0x1, sspace_mask 0x0000000000000003
NR5GML1   	nr5g_ml1_rrc_intf.c:            RRC-ML1 SPS config validate: cc 0, bwp_id: 0, is_bwp_config_sps: 0, RRC OTA cs_rnti_enable: 0, sps_cfg_active_bmask: 0, sps_cfg_rel_bmask: 0, sps_cfg_addmod_bmask: 0
NR5GML1   	nr5g_ml1_rrc_intf.c:            RRC-ML1 config validate:cc_id 0 bwp_id 0 BWP Cfg_mask 0x2|0x1- bwp_cmn_cfg 0x0|0x3, bwp_ded_cfg 0x3|0x0 
NR5GML1     nr5g_ml1_rrc_intf.c             ???
NR5GML1   	nr5g_ml1_rrc_intf.c:            RRC-ML1 config validate: BWP pdsch ded cfg validation
NR5GML1   	nr5g_ml1_rrc_intf.c:            RRC-ML1 config validate: ch_bw_used 0,ch_offset 0, carr_offset 0, bwp_start: 0, bwp_bw:52, bwp_end: 52, loc_and_bw 14025 for bwp_id 0
NR5GML1     nr5g_ml1_srch_api.c:            ???
NR5GML1     nr5g_ml1_srch_csi_rs.c:         ???
NR5GML1     nr5g_ml1_srch_csi_rs.c:         ???
**NR5GML1   	nr5g_ml1_rrc_intf.c:            RRC-ML1 config validate: Validation failed on CSI-RS params**
NR5GML1   	nr5g_ml1_rrc_intf.c:            RRC-ML1 cell group validate: Valid 0
NR5GRRC   	nr5g_rrc_llcdb.c:	            LLCDB: RRC API Validation failed: cphy_dedicated_cfg_req
NR5GRRC   	nr5g_rrc_utils.c:	            PROF_LOG: Did not find any entry time logged for this function call. Ignore func exit
NR5GRRC   	nr5g_rrc_llc.c:	                failed to construct lower layer cmds (validation failure). step1_step2_status(0x1)
 . . .
NR5GRRC     nr5g_rrc_llc.c:                 ???
NR5GRRC     nr5g_rrc_llc.c:                 ??? a0="L0: RRC Setup Delay                               :";a1=3720477028(0xDDC1F964);a2=18254(0x474E);a3=950(0x3B6);
NR5GRRC     nr5g_rrc_llc.c:                 ???	
 . . .
NR5GRRC   	nr5g_rrc_llc.c:                 LLCDB: cfg_reqi processing for proc (3) type (0) failed ! step1_step2_status (1) 
NR5GRRC   	nr5g_rrc_llcdb.c:               LLCDB: Revert L2 config in OC_ptr from backup
NR5GRRC     nr5g_rrc_llcdb.c:               ???
NR5GRRC   	nr5g_rrc_llcdb.c:               LLCDB: NR5G_RRC_CFG_CONN_REL/NR5G_RRC_CFG_ABORT_ML1. Init all cfg_masks, addmod_masks and active_masks
```

andrepuschmann, 2023-03-12:
Hey @AndrewOG255,
We are on this issue and will provide a patch shortly. Is it possible for you to switch to the new srsRAN Project?
Also, may I ask how you get those logs from the phone?

Thanks

AndrewOG255, 2023-03-13:
Hi @andrepuschmann,
 - Thanks for fixing this problem so we can move on
 - Yes, we would like to switch to srsRAN Project but for now we have one big impediment here: we need full documentation for all the options of config files for new srsRAN. The "documentation" may look like a commented example of some option that can be used.
 - You can get same logs with QXDM5 tool if you know how to configure the logging details level. But we use the other 3rd party tool that is not sold publicly. It just has a bit more convenient GUI but same logs and functionality.

AndrewOG255, 2023-03-13:
... configuration files:
[enb.conf.txt](https://github.com/srsran/srsRAN_4G/files/10952583/enb.conf.txt)
[rr.conf.txt](https://github.com/srsran/srsRAN_4G/files/10952585/rr.conf.txt)
[sib.conf.txt](https://github.com/srsran/srsRAN_4G/files/10952586/sib.conf.txt)
[rb.conf.txt](https://github.com/srsran/srsRAN_4G/files/10952587/rb.conf.txt)

suttonpd, 2023-03-13:
> * our company is interested to purchase srsRAN commercial license and use paid technical support and stack modifications to get the necessary things done if this possible ... how to contact you better?

Hi @AndrewOG255 , I can help with this. Please reach out to info(at)srs.io

ismagom, 2023-03-26:
> Hi @andrepuschmann,
>
> * Thanks for fixing this problem so we can move on
> * Yes, we would like to switch to srsRAN Project but for now we have one big impediment here: we need full documentation for [...]

Just a reminder that the configuration file reference documentation for srsRAN Project is already available: https://docs.srsran.com/projects/project/en/latest/user_manuals/source/config_ref.html

AndrewOG255, 2023-03-29:
@ismagom I will test it soon. Does it have CSI-RS problem fixed?

ismagom, 2023-03-29:
> @ismagom I will test it soon. Does it have CSI-RS problem fixed?

@AndrewOG255 it should, but would be great if you could test it.

pricalinat, 2023-03-30:
I have encountered the same problem. Has this problem been resolved on srsRAN_4G?

andrepuschmann, 2023-03-30:
@pricalinat There are currently now plans to backport the fixes we made to the srsrRAN_4G repo. We are kindly requesting all 5G-SA users to move to the new repository if possible.

pricalinat, 2023-03-31:
> @pricalinat There are currently no plans to backport the fixes we made to the srsrRAN_4G repo. We are kindly requesting all 5G-SA users to move to the new repository if possible.

Thank you for your reply. If you were asked to recommend a suitable COTS UE, which one would you recommend? Because OnePlus 10 Pro is in the recommended list in the document, but this problem still occurs after we buy it. We need a mobile phone that can be adapted to your project.

andrepuschmann, 2023-03-31:
We'll publish a list of tested phones shortly. But I can assure you that the OnePlus 10 Pro is working with the new code base.

pricalinat, 2023-04-03:
> We'll publish a list of tested phones shortly. But I can assure you that the OnePlus 10 Pro is working with the new code base.

Thank you for your response, but during the testing process, I found that my Oneplus 10 Pro cannot even find the network for the new project. In addition, I saw that you are using the Moto 30 Pro on Twitter. Is this phone now compatible? If so, I would consider buying it. Thanks again.

andrepuschmann, 2023-04-03:
The Moto Edge 30 Pro uses the exact same baseband as the Oneplus 10 Pro. Also the OnePlus 10 Pro has been reported to be working just fine. So the issue you're having with your setup is different.

andrepuschmann, 2023-04-25:
@AndrewOG255 I can't find a related issue in the new repo. Please consider to open one. For your tests, could you try to set the `pci` in `cell_cfg` back to `1`. We've found a regression that shows worse performance with some phone when the PCI is changed. This issue will be fixed in the next release (will come shortly) but it is still present in the current public repo.