srsran/srsRAN_4G #1110: DOWNLINK isn't working in SRSRAN-magma 5G setup

mounika-alavala, 2023-03-03:
Hi
We setup Magma AGW and orchestrator on Openstack VMs. We also setup SRSRAN on another openstack VM. We have used srsRAN and UERANSIM with MAGMA Core setup. srsRAN is able to integrate with magma. We are able to ping UE from AGW and vice versa. With SRSRAN, using iperf UPLINK is working but not DOWNLINK. UERANSIM is not getting attached, there is some problem with AMF Auth key(even after using hexadecimal).

Attached are the screenshots of the same.
Ping works:
[image]
[image]
Also, using iperf3, I am able to run traffic from the UE to agw (uplink) and see the trace on srsenb console.
[image]
[image]
[image]
But am not able to run traffic from agw to UE.
[image]
[image]
[image]
No firewall is running.
[image]
[image]

Can someone suggest a possible solution?
Thanks in advance.

andrepuschmann, 2023-03-03:
If ping works, both UL and DL is working. If your iperf3 DL session isn't working maybe verify the listening IPs etc. Maybe also try iperf and use UDP to confirm the traffic is flowing there.

DeepakalaB, 2023-03-03:
UE is attached successfully and assigned IP - 192.168.128.21. 
iperf server is listening on UE side
[image]
Ping works from agw to UE 
[image]
But we see no traffic in the iperf DL session:
[image]
[image]
The following screenshots  were taken of iperf3 DL sessions:
AGW side:
[image]
UE side:
[image]
[image]

Any help is appreciated. Thanks!