srsran/srsRAN_4G #1111: Compilation Errors Using srsRAN_4G 22.10.0

chaohwa4, 2023-03-07:
## Issue Description ##
There are compiling errors while compiling srsRAN 4G Version 22.10.0

## Setup Details ##
Analog Devices PackRF (AD9361/Z7035) with multiple RF channels. Analog Devices Kuiper Linux 2021_r1, based on Raspberry Pi OS 11 (Bullseye) with Linux Kernel 5.10 and GCC 10.2.1.

## Expected Behavior ##
Successful compilation.

## Actual Behaviour ##
The error message was shown in the following file. [cmake_make_error_2021_R1.log](https://github.com/srsran/srsRAN_4G/files/10910180/cmake_make_error_2021_R1.log)

## Steps to reproduce the problem ##
1. Build new SD card based on 2021_R1
2. Type "sudo adi_update_tools.sh" for update
3. Install the following: 
    sudo apt-get install libfftw3-dev
    sudo apt-get update
    sudo apt-get install libmbedtls-dev
    sudo apt-get install build-essential cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev
    sudo apt-get install libdw-dev
    sudo apt-get install libzmq-dev
    sudo apt-get install soapysdr-module-all soapysdr-tools libsoapysdr-dev
    sudo apt-get install libsctp-dev lksctp-tools 
    sudo apt-get install libczmq-dev
4. Download Soapy Pluto SDR "git clone https://github.com/pothosware/SoapyPlutoSDR"
5. Build and install Soapy Pluto SDR
    cd SoapyPlutoSDR
    mkdir build
    cd build
    cmake ../
    make
    make install    
6. Download SRS RAN "git clone https://github.com/srsRAN/srsRAN_4G.git"
7. Build and install SRS RAN 
    cd srsRAN_4G
    mkdir build
    cd build
    cmake ../
    make

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

ninjab3s, 2023-03-24:
After a quick check I have seen that the Analog Devices PackRF (AD9361/Z7035) has only 1GB of ram. Is this correct? I think the build fails due to having too less RAM. I also think that its not possible to run srsRAN 4G with that less RAM.

chaohwa4, 2023-03-24:
Yes, the PackRF has only 1GB of RAM.

chaohwa4, 2023-03-24:
The weird thing is that when I compiled srsRAN_4G using Analog Devices Kuiper Linux 2019_r2, based on Raspberry Pi OS 10 (Buster) with Linux Kernel 4.19 and GCC 8.3, the build was compiled successfully.

ninjab3s, 2023-03-24:
Maybe this was due to the different GCC version or an older commit. To confirm my assumption from above you can check the output of `dmesg`. This should show if gcc went OOM.

> The weird thing is that when I compiled srsRAN_4G using Analog Devices Kuiper Linux 2019_r2, based on Raspberry Pi OS 10 (Buster) with Linux Kernel 4.19 and GCC 8.3, the build was compiled successfully.

Were you able to run srsRAN on that platform?

chaohwa4, 2023-03-24:
I could run srsRAN UE in that platform, but n_prb has to be 6 or 15. Above that the process will get killed due to memory limit.

ninjab3s, 2023-03-24:
Could you verify that the limited RAM was the issue? If this is the case maybe you can cross compile srsRAN on another system. Otherwise, is installing srsUE via apt an option that could work for you? [1]

[1] https://docs.srsran.com/projects/4g/en/latest/general/source/1_installation.html#package-installation

chaohwa4, 2023-03-24:
Here are the results by package installation
root@analog:~# sudo add-apt-repository ppa:softwareradiosystems/srsran_4g
Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 95, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 109, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 599, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 91, in get_sources
    raise NoDistroTemplateException(
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Raspbian/bullseye
root@analog:~# sudo apt-get install srsran_4g -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package srsran_4g

ninjab3s, 2023-04-03:
My bad, we dont provide binaries for arm. I think the best option you have is to cross compile or compile on arm hardware like a Raspberry Pi and copy the binary to your system.

chaohwa4, 2023-04-08:
I tried gcc-8 and gcc-9 and they did not work, either.

The conclusion is that do not use Analog Devices Kuiper Linux 2021_r1, based on Raspberry Pi OS 11 (Bullseye) at this time and use Analog Devices Kuiper Linux 2019_r2, based on Raspberry Pi OS 10 (Buster) to compile SRS RAN.

closed 2023-04-08 by chaohwa4