srsran/srsRAN_4G #1116: release_22_10 - compile error

boaks, 2023-03-12:
Ubuntu 20.04, 
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1) 

steps according [installation - Download and build srsRAN 4G:](https://docs.srsran.com/projects/4g/en/latest/general/source/1_installation.html):

```
git clone https://github.com/srsRAN/srsRAN_4G.git
cd srsRAN_4G
git checkout release_22_10
mkdir build
cd build
cmake ../
make
```

> In file included from /home/achim/repos/srsRAN_4G/srsenb/hdr/stack/mac/sched_phy_ch/../sched_lte_common.h:26,
>                  from /home/achim/repos/srsRAN_4G/srsenb/hdr/stack/mac/sched_phy_ch/sched_result.h:25,
>                  from /home/achim/repos/srsRAN_4G/srsenb/hdr/stack/mac/sched_grid.h:27,
>                  from /home/achim/repos/srsRAN_4G/srsenb/src/stack/mac/sched_grid.cc:22:
> /home/achim/repos/srsRAN_4G/lib/include/srsran/adt/bounded_bitset.h: In member function ‘srsenb::alloc_result srsenb::sf_grid_t::alloc_ul_data(srsenb::sched_ue*, srsenb::prb_interval, bool, bool)’:
> /home/achim/repos/srsRAN_4G/lib/include/srsran/adt/bounded_bitset.h:336:32: error: ‘newmask.srsran::bounded_bitset<100, true>::buffer[<unknown>]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
>   336 |       buffer[i] &= other.buffer[i];
>       |                    ~~~~~~~~~~~~^
> cc1plus: all warnings being treated as errors
> make[2]: *** [srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/build.make:115: srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/sched_grid.cc.o] Fehler 1
> make[1]: *** [CMakeFiles/Makefile2:12473: srsenb/src/stack/mac/CMakeFiles/srsenb_mac.dir/all] Fehler 2
> make: *** [Makefile:163: all] Fehler 2
> 

The same error occurs on the current master commit.

boaks, 2023-03-12:
Compiles with 

gcc version 10.3.0 (Ubuntu 10.3.0-1ubuntu1~20.04) 

Maybe more a compiler error?

ninjab3s, 2023-03-24:
Hello boaks,
I was able to compile srsRAN_4G on `release_22_10` on Ubuntu with GCC `Ubuntu 9.4.0-1ubuntu1~20.04.1` without errors. I assume that some of the dependencies could cause this issue. Since you have found a solution, can we close the issue?

boaks, 2023-03-24:
> I was able to compile srsRAN_4G on release_22_10 on Ubuntu with GCC Ubuntu 9.4.0-1ubuntu1~20.04.1 without errors.

interesting.

> Since you have found a solution, can we close the issue?

yes, sure.

closed 2023-03-24 by boaks