srsran/srsRAN_4G #1120: 5G SA End-to-End - Ping does not work

alejandroBlancoPizarro, 2023-03-15:
## Issue Description ##
Thanks so much for the effort srsRAN_4G team.

I am following this tutorial: https://docs.srsran.com/projects/4g/en/latest/app_notes/source/5g_sa_E2E/source/index.html

So far everything is fine. But after connecting the srsue, the ping does not work. 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
Ubuntu 22.04, latest version srsRAN_4G, open5gs, ZMQ

## Expected Behavior ##
After attaching the srsue, I expect the ping to work. 

## Actual Behaviour ##
Ping does not work. sudo ip netns exec ue1 ping 10.45.0.1
[image]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
I follow this tutorial:  https://docs.srsran.com/projects/4g/en/latest/app_notes/source/5g_sa_E2E/source/index.html

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
[image]

There are some errors in open5gs once I run the ping
[image]

In addition, I cannot see the interface for the open5gs core which is ogstun:

[image]

LOGS are found below.
[logs.zip](https://github.com/srsran/srsRAN_4G/files/10979222/logs.zip)

ninjab3s, 2023-03-24:
Looking at your logs I see some errors in the UPF in open5gs. I assume that this is causing the actual problem. From `log_core.txt`:

`03/15 11:27:20.516: [upf] WARNING: PFCP[RSP] has already been associated [127.0.0.4]:8805 (../src/upf/pfcp-sm.c:213)`
`03/15 11:27:27.517: [sock] ERROR: ogs_write() failed (5:Input/output error) (../lib/tun/tunio.c:83)`

I think fixing these issues should solve your problem.

ismagom, 2023-06-01:
Please use srsRAN_Project for 5G SA

closed 2023-06-01 by ismagom