srsran/srsRAN_4G #1122: How do I transfer real files via srsLTE?

JunzheCheng, 2023-03-17:
Hello,

I was trying to simulate srsLTE and I ran into some issues.

1. According to the srsRAN documentation, srsEPC and srsENB work on the same computer by default, and the documentation mentions that they can work separately on two computers, how to make them work on two computers?

2. How do I transfer real files via srsLTE? Through the srsLTE documentation, after we built the complete architecture of srsUE-ENB-EPC, we found that the connection between UE and ENB and EPC can be confirmed by ping the computer successfully. But how do I transfer other files through the connection?

Looking for your replies! Your help would be appreciated.

ninjab3s, 2023-03-24:
1. Please refer to https://docs.srsran.com/projects/4g/en/latest/usermanuals/source/srsepc/source/2_epc_getstarted.html#enbs-and-srsepc-on-separate-machines

2. This depends on the software you can run on the EPC and UE side. If `ssh/scp` are available this could be an option.

andrepuschmann, 2023-04-25:
@Egwel See https://docs.srsran.com/projects/4g/en/latest/app_notes/source/zeromq/source/index.html

The same also works if the radio is real, i.e. not ZMQ.

Closing issue.