srsran/srsRAN_4G #1137: Connection between UE and ENB lost after several seconds

jqhu27, 2023-03-29:
## Issue Description ##
[Describe the issue in detail]
srsUE can attach to the network at the first several seconds and the assigned IP address can be seen from the terminal. But after several seconds, the connection is lost and I cannot ping srsUE from srsEPC or vice versa.
[image]
[image]
[image]

I use the modified srsRAN by POWDER. The modified srsRAN is integrated with E2 terminal which is then used to establish ORAN testbed. I followed the OAIC to deploy the ORAN testbed [The](https://github.com/openaicellular/oaic). Some people succeed following the instructions but I failed. I am wondering whether this problem is resulting from srsRAN side. 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
Operation System: Ubuntu 20.04
Hardware: Dell Latitude3420
RF front-end: x310
UHD versions: UHD_4.1.0.6-120-g7e1ff96a

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2023-03-29:
To me that looks like a normal release due to inactivity. And then the UE attaches again when there is data to send.

closed 2023-03-29 by andrepuschmann

jqhu27, 2023-03-29:
Hi thanks for your information. But when I try to ping srsUE from srsEPC, there is no information.

andrepuschmann, 2023-03-29:
Two thing to suggest: first I would repeat the test with vanilla srsRAN to see if the issues persists. If not, the likely it's related to the modifications done and so it's probably better to ask the authors.

If the issue still exists with vanilla srsRAN then please send full logs of all components so we at least have the chance to help.

jqhu27, 2023-03-30:
Hi, thanks for getting back to me. I attached the UE and ENB log file below. Could you please take a look at it? Thank you in advance.
[epc.log](https://github.com/srsran/srsRAN_4G/files/11113383/epc.log)
[ue.log](https://github.com/srsran/srsRAN_4G/files/11113384/ue.log)