srsran/srsRAN_4G #1139: SrsGUi Question

psi7, 2023-04-02:
I am trying to trun srsGui after building and setting up srsGui and srsRan.
When it try to run srsenb and I want to have the gui plots I am getting:

> Trying to start a plot but plots are disabled ( ENABLE_GUI constant in sf_worker.cc)

 What should I do exaclty? I have been trying to fix it for a week.

CatiaS11, 2023-04-05:
Hello ! 
You should go to root/.conf/srsend -> then when you edit the file you'll see a section called [GUI] and you have to enable GUI option there. I worked for me in any case. Check if you built srsGUI and srsRAN properly.

CatiaS11, 2023-04-05:
srsenb*

psi7, 2023-04-05:
i am running the command : srsenb --gui.enable=yes and it works now but I am getting the four basic diagrams (Channel Response Magnitude, PUSCH-Equalized Symbols, Channel Response Argument, PUCCH- Equalized Symbols) with the above command. Am I missing something? Are there other commands or diagrams that SRS can give me?

CatiaS11, 2023-04-05:
> Thanks for the immediate response. I am running the command : srsenb --gui.enable=yes and it works now but I am getting the four basic diagrams (Channel Response Magnitude, PUSCH-Equalized Symbols, Channel Response Argument, PUCCH- Equalized Symbols) with the above command. Am I missing something? Are there other commands or diagrams that srsGUI can give me?

You can have only four diagrams I guess.

xavierarteaga, 2023-04-27:
Hi @psi7,
@CatiaS11 is correct. Currently, only graphs are implemented.

You could potentially implement your plots by looking at the current examples.

I am closing this issue. Feel free to reopen if you have more questions.

xavierarteaga, 2023-04-27:
Hi @psi7,
@CatiaS11 is correct. Currently, only these four graphs are implemented.

You could potentially implement your plots by looking at the current examples.

I am closing this issue. Feel free to reopen if you have more questions.

closed 2023-04-27 by xavierarteaga