srsran/srsRAN_4G #1140: srsRAN_4G end to end using usrp x310 - attach failed

xuebashuoge, 2023-04-04:
## Issue Description ##

I am trying to establish an end-to-end 4G network and my UE cannot attach to eNB. I am using the default configuration files and I could not find what is the cause of this problem.

## Setup Details ##
2PC + 2 usrp x310
Ubuntu 20.04, latest srsRAN_4G v22.10.0, uhd v3.15.0.0
PC1 - running the UE with the default ue.conf
PC2 - running the EPC with the default epc.conf and eNB with the default enb.conf

## Expected Behavior ##
I want to connect PC1 to PC2 through 4G network. I can ping PC1 on PC2 and ping PC2 on PC1.

## Actual Behaviour ##
UE  message show attached failed and radio link failure
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_soapy.so libsrsran_rf_zmq.so                                                                                                                                                                                                
Inactive RF plugins:                                                                                                                                                                                                                                                            
Reading configuration file ue.conf...                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                                                
Built in Release mode using commit 00c972ac4 on branch master.                                                                                                                                                                                                                  
                                                                                                                                                                                                                                                                                
Opening 1 channels in RF device=default with args=default                                                                                                                                                                                                                       
Supported RF device list: UHD soapy zmq file                                                                                                                                                                                                                                    
Trying to open RF device 'UHD'                                                                                                                                                                                                                                                  
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_3.15.0.0-0-unknown                                                                                                                                                                                                 
[INFO] [LOGGING] Fastpath logging disabled at runtime.                                                                                                                                                                                                                          
Opening USRP channels=1, args: type=x300,master_clock_rate=184.32e6                                                                                                                                                                                                             
[INFO] [UHD RF] RF UHD Generic instance constructed                                                                                                                                                                                                                             
[INFO] [X300] X300 initialization sequence...                                                                                                                                                                                                                                   
[INFO] [X300] Maximum frame size: 8000 bytes.                                                                                                                                                                                                                                   
[INFO] [X300] Radio 1x clock: 184.32 MHz                                                                                                                                                                                                                                        
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)                                                                                                                                                                                                    
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1302 MB/s)                                                                                                                                                                                                                        
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1323 MB/s)                                                                                                                                                                                                                        
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)                                                                                                                                                                                                      
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)                                                                                                                                                                                                      
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)                                                                                                                                                                                                        
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)                                                                                                                                                                                                        
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)                                                                                                                                                                                                        
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)                                                                                                                                                                                                        
RF device 'UHD' successfully opened                                                                                                                                                                                                                                             
Waiting PHY to initialize ... done!                                                                                                                                                                                                                                             
Attaching UE...                                                                                                                                                                                                                                                                 
.                                                                                                                                                                                                                                       [... cut ...]
RACH:  tti=7111, cc=0, pci=1, preamble=48, offset=1, temp_crnti=0x4a                                     
Disconnecting rnti=0x49.

ue.log
[image]

eNB.log
[image]

epc.log
[image]

## Steps to reproduce the problem ##
I use the default configuration files generated by srsran_install_configs.sh. I open the terminal in the folder that contains the default *.conf on both PC. I use the command ```sudo srsepc epc.conf``` for EPC, ```sudo srsenb enb.conf``` for eNB, and ```sudo srsue ue.conf``` for UE.

bigbossolidsnake, 2023-04-05:
I think i am facing similar attach issues as you with srsRAN 4G 22.10  with different RF front end. Except in my case i am getting a very brief attach but soon detaches. You are using USRP , i am using LimeSDR USB. See [here](https://github.com/srsran/srsRAN_4G/issues/1141)

nedsana, 2023-05-10:
Hi, same problem here. I use the master branch of release_22_10 (commit 254cc719a) with LimeSDR-Mini and antennas. 
I am trying to connect 2 srsue instances to srsenb using smaller bandwidth (n_prb = 15), in order to save CPU resources. But this does not seem to work. 

My setup uses:
- PC (i7-7700K 8 cores/8 threads), which runs srsepc, srsenb and _**one srsue**_. 
- Laptop (i7-9850H 6 cores/12 threads), which runs the second srsue

I found that only if I set n_prb = 100 in enb.conf, then the srsue, running on the PC is likely to connect. The srsue on the laptop, rarely establishe RRC connection, but never reaches the EPC stack. Below you can see the STDOUT.

> UE0 (PC):
Found Cell:  Mode=FDD, PCI=1, PRB=100, Ports=1, CP=Normal, CFO=-1.8 KHz
RF status: O=0, U=0, L=1
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 23.04 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=22, tti=7381, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=14
Network attach successful. IP: 172.16.0.2
LSoftware Radio Systems RAN (srsRAN) 10/5/2023 13:34:54 TZ:0
RF status: O=0, U=0, L=1
Warning: Detected Radio-Link Failure
LRRC Connection Reestablishment to PCI=1, EARFCN=1100 (Cause: "otherFailure")
RF status: O=0, U=0, L=1
Random Access Transmission: seq=31, tti=7811, ra-rnti=0x2
LRandom Access Transmission: seq=29, tti=7831, ra-rnti=0x2
LRandom Access Transmission: seq=8, tti=7851, ra-rnti=0x2
LRandom Access Transmission: seq=26, tti=7871, ra-rnti=0x2
LRandom Access Transmission: seq=28, tti=7891, ra-rnti=0x2
Timer T301 expired: Going to RRC IDLE
RRC IDLE
RF status: O=0, U=0, L=4
Service Request with cause mo-Data.
LRandom Access Transmission: seq=46, tti=7941, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x47, ta=14
Service Request successful.
RF status: O=0, U=0, L=1

> UE1 (Laptop):
Found Cell:  Mode=FDD, PCI=1, PRB=100, Ports=1, CP=Normal, CFO=-0.2 KHz
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 23.04 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
RF status: O=0, U=0, L=1
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=27, tti=6261, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x48, ta=14
LLLRF status: O=0, U=0, L=3
LWarning: Detected Radio-Link Failure
RRC IDLE
LRF status: O=0, U=0, L=2
LLRF status: O=0, U=0, L=2
LRF status: O=0, U=0, L=1
Attach failed (attempt 1/5)
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
L.
Found Cell:  Mode=FDD, PCI=1, PRB=100, Ports=1, CP=Normal, CFO=-0.4 KHz
RF status: O=0, U=0, L=3
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 23.04 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
RF status: O=0, U=0, L=1
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=32, tti=2261, ra-rnti=0x2
LRandom Access Transmission: seq=25, tti=2281, ra-rnti=0x2
LRandom Access Transmission: seq=29, tti=2301, ra-rnti=0x2
LRandom Access Transmission: seq=27, tti=2321, ra-rnti=0x2
LRandom Access Transmission: seq=48, tti=2341, ra-rnti=0x2
LRandom Access Transmission: seq=15, tti=2361, ra-rnti=0x2
LRandom Access Transmission: seq=48, tti=2381, ra-rnti=0x2
LRandom Access Transmission: seq=41, tti=2401, ra-rnti=0x2
LRandom Access Transmission: seq=41, tti=2421, ra-rnti=0x2
LRandom Access Transmission: seq=33, tti=2441, ra-rnti=0x2
RF status: O=0, U=0, L=9

> ENB:
Setting frequency: DL=1980.0 Mhz, UL=1900.0 MHz for cc_idx=0 nof_prb=100
[INFO] Tx calibration finished
[INFO] Rx calibration finished
RACH:  tti=7381, cc=0, pci=1, preamble=22, offset=14, temp_crnti=0x46
User 0x46 connected
LDisconnecting rnti=0x46.
RACH:  tti=7941, cc=0, pci=1, preamble=46, offset=14, temp_crnti=0x47
User 0x47 connected
RACH:  tti=6261, cc=0, pci=1, preamble=27, offset=14, temp_crnti=0x48
Disconnecting rnti=0x48.

ismagom, 2023-06-01:
Hi @xuebashuoge, could you attach logs of both sides in info level please?

@nedsana, you can't run the UE and eNB in the same PC, sorry

xuebashuoge, 2023-06-02:
> 

I have show the logs (ue.log, enb.log, epc.log) in my post, we can refer to those photos.

nedsana, 2023-06-02:
Hi ismagom, 

Yes, I have read that ENB and UE are recommended to run on separate machines. 

But I have a limited setup: 2 PCs and 4 SDRs. On the first PC I run ENB and 2 UEs. On the second PC I run one UE.
Also for the moment I am only interested in signalling, not data transfer. 

So I decided to give it a try. 

To do so I replaced the antennas, which I used in my previous post, with cables and splitters. 

And as I have described above, only if I set n_prb=100 or n_prb=50 in enb.conf and compile the binaries in Release mode, I successfully connect all the UEs to the ENB. OK, I admit this is not 100% success. Now and then some UE fails to connect. But this is not frequent, so for me this is usable.

The problem is that with any value, different from n_prb=100 or n_prb=50, the UEs do not connect. 
No matter if the UEs runs on the same PC or not.

I have tested the same configuration with ZMQ, instead of SDRs, where I use n_prb=15. And there all works great - all UEs connect.

It seems there is something wrong with the radio data processing for n_prb=6,15,25,75. 
But I am not familiar with the code and I have no guidelines for debugging.

Greetings.