srsran/srsRAN_4G #1145: Decoding capabilities with half-duplex SDRs

un-kasimir, 2023-04-10:
## Issue Description ##

Hey! I'm working on a university project where we want to use very low cost SDRs (like the HackRF One) to capture information about the LTE cell and it's nodes (occupancy of the cell for example).

I hope you guys are not annoyed by such requests - I've read requests like #320 and #287 which were about using the HackRF with srsRAN.

But, since you don't explicitly say that srsRAN supports only full-duplex hardware on [this](https://docs.srsran.com/projects/4g/en/latest/app_notes/source/hw_packs/source/index.html) page, I assume it is okay to ask this here:

I'm currently evaluating how to retrieve information from a UE perspective about the current state of LTE cells in the area. I hope I can do this only by reading information without actively transmitting messages (I assume after passing the synchronization sequence, this should be possible in general). 

Therefore, I thought the srsRAN 4G library might help me decoding messages and retrieving such information. So, without having a closer look at your code, can you tell me whether it might be possible to make use of your decode capabilities by using a half-duply-only SDR?

Thanks for your awsome work in advance.

Pinging @andrepuschmann since you commented on such questions before :slightly_smiling_face:

andrepuschmann, 2023-04-10:
My earlier comment about the HackRF not being suitable for LTE isn't quite right. Let me explain. For a full LTE UE or eNB, unless it's TDD, you do need full-duplex capabilities. However, if you're doing TDD only and you have the right mechanisms in place to switch between Rx and Tx in a timely manner, you might be able to use half-duplex HW. Before you ask - we don't have full TDD support in srsRAN 4G so it hasn't been tested at all.

The other area of application of half-duplex HW such as the HackRF would be Rx/Tx-only use-cases of course. I've never tried it but I am pretty sure that the HackRF works with the `pdsch_ue/enodeb` pretty well. Your best bet is probably the SoapySDR RF driver with the HackRF module.

un-kasimir, 2023-04-24:
Thanks a lot for your quick response and the information!

I'll keep you updated if I get this running with half-duplex hardware.

closed 2023-04-24 by un-kasimir