srsran/srsRAN_4G #1147: srsRAN <-- > Aether-In-A-Box (OMEC 4G-Core ) --> Getting Network failure - Code 17 | 19 | 66

MD-star-cloud33, 2023-04-11:
I've 2 VM, in 1st VM Aether-in-a=Box 4G Core Running as a CNF & in 2nd VM srsRAN running with eNB & UE with ZMQ.
Changed ~/.config/srsran/enb.conf
[enb]
enb_id = 0x19B
mcc = 208
mnc = 01
mme_addr = X.X.X.X (My Node Pvt IP Address where Core component Running )
gtp_bind_addr = X.X.X.X (srsRAN VM own IP)
s1c_bind_addr = X.X.X.X (srsRAN VM own IP)
s1c_bind_port = 36412 (port for SCTP)
n_prb = 50

Changed ~/.config/srsran/ue.conf

[usim]
mode = soft
algo = milenage
opc = d4416644f6154936193433dd20a0ace0 (As per HSS)
k = 465b5ce8b199b49faa5f0a2ee238a6bc (As per HSS)
imsi = 208010167891201 (As per HSS)
imei = 353490069873319
#reader =
#pin = 1234

[nas]
apn = internet
apn_protocol = ipv4

## Issue Description ##
Downlink NAS Transport Attach Released.

## Setup Details ##
Taken 2 VM in AWS, each of these vm are reachable from one another & from internet,

## Expected Behavior ##
e2e call should progress

## Actual Behaviour ##
ubuntu@ip-172-31-3-99:~/srsRAN_4G/build$ sudo ./srsenb/src/srsenb /home/ubuntu/.config/srsran/enb.conf --rf.device_name=zmq --ce_args="fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=23.04e6"
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins:
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/ubuntu/.config/srsran/enb.conf...

Built in Release mode using commit 5a4eb5d6c on branch master.

Opening 1 channels in RF device=zmq with args=fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enbrate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=enb
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2000
CH0 fail_on_disconnect=true

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
t
Enter t to stop trace.
RACH:  tti=341, cc=0, pci=1, preamble=45, offset=0, temp_crnti=0x46

               -----------------DL----------------|-------------------------UL-------------------------
rat  pci rnti  cqi  ri  mcs  brate   ok  nok  (%) | pusch  pucch  phr  mcs  brate   ok  nok  (%)    bsr
lte    1   46   15   0    9   6.5k   16    8  33% |  99.9   99.9   30   16    41k    8    0   0%    0.0
lte    1   46   15   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
lte    1   46   15   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
lte    1   46   15   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
lte    1   46   15   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
Disconnecting rnti=0x46.

UE Status
ubuntu@ip-172-31-3-99:~/srsRAN_4G/build$ sudo ./srsue/src/srsue /home/ubuntu/.config/srsran/ue.conf --rf.device_name=zmq --rf.args="tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6" --gw.netns=ue1
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins:
Reading configuration file /home/ubuntu/.config/srsran/ue.conf...

Built in Release mode using commit 5a4eb5d6c on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Waiting PHY to initialize ... done!
Attaching UE...
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.2 KHz
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Found PLMN:  Id=20801, TAC=1
Random Access Transmission: seq=45, tti=341, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=0
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.2 KHz
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Found PLMN:  Id=20801, TAC=1
Attach failed (attempt 1/5)
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.2 KHz
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Found PLMN:  Id=20801, TAC=1
Attach failed (attempt 2/5)

MME Log & SPGW-C log also attached.

Getting error ->
Cause https://github.com/srsran/srsRAN_4G/issues/16 – MSC temporarily not reachable

  This EMM cause is sent to the UE if it requests a combined EPS attach or tracking area updating in a PLMN where the MSC is temporarily not reachable via the EPS part of the network.
Cause https://github.com/srsran/srsRAN_4G/issues/17 – Network failure

  This EMM cause is sent to the UE if the MME cannot service an UE generated request because of PLMN failures.
Cause https://github.com/srsran/srsRAN_4G/pull/66 – Requested APN not supported in current RAT and PLMN combination

  This ESM cause is used by the network to indicate that the procedure requested by the UE was rejected as the requeste
## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

[PCAP-mme-spgwc-log.zip](https://github.com/srsran/srsRAN_4G/files/11196622/PCAP-mme-spgwc-log.zip)

[enb.log.zip](https://github.com/srsran/srsRAN_4G/files/11196630/enb.log.zip)
[ue.log.zip](https://github.com/srsran/srsRAN_4G/files/11196694/ue.log.zip)
[mme-app.txt](https://github.com/srsran/srsRAN_4G/files/11196696/mme-app.txt)
[s1ap-app.txt](https://github.com/srsran/srsRAN_4G/files/11196697/s1ap-app.txt)
[spgwc.txt](https://github.com/srsran/srsRAN_4G/files/11196698/spgwc.txt)
[aiab-1104.zip](https://github.com/srsran/srsRAN_4G/files/11196701/aiab-1104.zip)

MD-star-cloud33, 2023-04-11:
if you could check these logs once in your free time. Have been changing configs in core setup from last 7-8 days, Please help.

infinitydon, 2023-05-01:
Also getting the same issue but looks like this is the in security capabilties exchange..

srsRAN is making mandatory **_(ESM information transfer): Security protected ESM information transfer required_** but MME is not relaying it back, so srsRAN is rejecting UE security capability

UE Attach request:

[image]

MME replayed security capabilities does not include **_ESM information transfer_**

[image]

Can ESM be made optional?

infinitydon, 2023-05-01:
Using latest MME release seems to fix the ESM problem, currently MME is responding with successful IP allocation but srsUE is still showing attach failed..

```
root@srslte-85cf55cc44-tcq24:/srsRAN/build/srsue/src# ./srsue ue.conf
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins:
Reading configuration file ue.conf...

Built in Release mode using commit 655a098b7 on branch master.

Opening 1 channels in RF device=zmq with args="tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6"
Supported RF device list: zmq file
CHx base_srate=23.04e6"
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Operation not permitted
Waiting PHY to initialize ... Warning: Failed to create thread with real-time priority. Creating it with normal priority: Operation not permitted
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Operation not permitted
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Operation not permitted
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Operation not permitted
done!
Attaching UE...
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.2 KHz
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Found PLMN:  Id=20801, TAC=1
Random Access Transmission: seq=22, tti=341, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=0
Attach failed (attempt 1/5)
```
[image]

Attached is the full PCAP

[srs-onf-attach-may-1-3.zip](https://github.com/srsran/srsRAN_4G/files/11366583/srs-onf-attach-may-1-3.zip)

infinitydon, 2023-05-01:
UE debug logs

[ue.zip](https://github.com/srsran/srsRAN_4G/files/11366696/ue.zip)