srsran/srsRAN_4G #1153: 5G SA COTS UE: Cannot find sim card when using pysim-shell

yzucyh, 2023-05-02:
## Issue Description ##

Hi all,
I'm practicing the project below
[official website](https://docs.srsran.com/projects/4g/en/latest/app_notes/source/5g_sa_COTS/source/index.html)
I've bought the programmable sim card and adapter, reader also. When I try to read the card it shows that (no card)
`eric@eric-ASUSPRO-D840MB-M840MB:~/Downloads/pysim$ ./pySim-shell.py -p 0`
`Using PC/SC reader interface`
`Waiting for card...`
`No card detected!`
`pySim-shell not equipped!`
`Welcome to pySim-shell!`
`pySIM-shell (no card)>`

I follow this guide to read the info, but I failed in the first step
[here](https://gist.github.com/mrlnc/01d6300f1904f154d969ff205136b753)
So, i'm wondering if there is a way to test my adapter to work fine?

THANKS FOR ALL.

## Setup Details ##
Ubuntu 22.04
Device: USRP B210
Programmable card: sysmoISIM-SJA2 
adapter: sysmocom Professional SIM card adapter
reader: sysmocom Omnikey CardMan 3121 USB CCID interface
[image]
Is there any problem with my setting? and I also tried 3ff/4ff still not work.

Thanks again.

## Expected Behavior ##
Expect to see my programmable sim card to add my adm

## Actual Behaviour ##
Show no card

yzucyh, 2023-05-02:
I've used `pcsc_scan` to check, I think it found my sim card, but I don't know why pysim can't read my sim card.
[image]

andrepuschmann, 2023-05-03:
This is not the `pysim` support forum. Better check there.

closed 2023-05-03 by andrepuschmann