srsran/srsRAN_4G #1154: trouble running srsGUI in SRSRAN 4G

leovassalo, 2023-05-02:
## Issue Description ##
I cannot seem to run the srsGUI, even though i am using the tag `--gui_enable=yes` when running the `sudo srsenb` command. I receive the following response:

`Trying to start a plot but plots are disabled (ENABLE_GUI constant in sf_worker.cc)`

I have already tried rebuilding both softwares.

## Setup Details ##
Ubuntu 20.04 with SRSRAN 4G 22.10
ENB with BladeRF A9

closed 2023-05-03 by andrepuschmann

leovassalo, 2023-05-04:
Greetings, thank you for your answer.

I have reinstalled srsRAN_4G after installing the srsGUI. However, I now get this output when trying to run the eNB:

```srsenb: error while loading shared libraries: libsrsgui.so: cannot open shared object file: No such file or directory```

Also, please take a look at my cmake output. Does it indicate any missing libraries regarding the gui? Because I believe it was properly identified here.

```
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_SYSTEM: Linux-5.15.0-67-generic
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Build type not specified: defaulting to Release.
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB - Success
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Enabling ccache for C
-- Enabling ccache for CXX
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.8
-- FFTW3F LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- FFT_LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Found MbedTLS: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so  
-- Checking for module 'libpcsclite'
--   No package 'libpcsclite' found
-- Could NOT find PCSCLite (missing: PCSCLITE_LIBRARY PCSCLITE_INCLUDE_DIR) 
-- PCSC LIBRARIES: PCSCLITE_LIBRARY-NOTFOUND
-- PCSC INCLUDE DIRS: PCSCLITE_INCLUDE_DIR-NOTFOUND
-- UHD LIBRARIES /usr/lib/x86_64-linux-gnu/libuhd.so
-- UHD INCLUDE DIRS /usr/include
-- Found UHD: /usr/lib/x86_64-linux-gnu/libuhd.so  
-- Performing Test UHD_ENABLE_X300_FW_RESET
-- Performing Test UHD_ENABLE_X300_FW_RESET - Failed
-- Performing Test UHD_ENABLE_RFNOC
-- Performing Test UHD_ENABLE_RFNOC - Success
-- Performing Test UHD_ENABLE_CUSTOM_RFNOC
-- Performing Test UHD_ENABLE_CUSTOM_RFNOC - Failed
[... 68 lines cut ...]
-- Building with srsENB/srsGNB
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
-- Configuring done
-- Generating done
-- Build files have been written to: /home/leovass2/srsRAN_4G/build

```

andrepuschmann, 2023-05-04:
the install path is not in your system library path.
try `export LD_LIBRARY_PATH=/usr/local/lib`

leovassalo, 2023-05-04:
Sorry, it did not work. Should I write this in any terminal? Under any folder? Is this command exactly what I need to write or should I change anything?

Sorry for my lack of knowledge.