srsran/srsRAN_4G #1160: ue earfcn list iteration issue

jgiovatto, 2023-05-04:
## Issue Description ##
setting ue.conf
dl_earfcn = 2850,3050

Does not work as expected.

## Expected Behavior ##
Expect to change frequency during cell search .

## Actual Behaviour ##
Uses the first entry only.

## Additional Information ##
Proposed fix:

'''
diff --git a/srsue/src/phy/sync.cc b/srsue/src/phy/sync.cc
index c346d2cf3..905aece2d 100644
--- a/srsue/src/phy/sync.cc
+++ b/srsue/src/phy/sync.cc
@@ -290,7 +290,7 @@ rrc_interface_phy_lte::cell_search_ret_t sync::cell_search_start(phy_cell_t* fou
   }
 
   cellsearch_earfcn_index++;
-  if (cellsearch_earfcn_index >= worker_com->args->dl_earfcn_list.size() or earfcn < 0) {
+  if (cellsearch_earfcn_index >= worker_com->args->dl_earfcn_list.size() or earfcn == 0) {
     Info("Cell Search: No more frequencies in the current EARFCN set");
     cellsearch_earfcn_index = 0;
     ret.last_freq           = rrc_interface_phy_lte::cell_search_ret_t::NO_MORE_FREQS;
'''

therustyalex, 2024-09-10:
Did you solve this?

stackprogramer, 2024-11-08:
Did anyone find a solution for changing parameters in real time?