srsran/srsRAN_4G #1163: srsRAN gNB with srsUE tun interface is not created

gustavobsch, 2023-05-06:
## Issue Description ##
[Describe the issue in detail]
I'm following the recently created 'srsRAN gNB with srsUE' user guide and when I run srsue it can register with the open5gs network fine, it gets a valid IP from the DN assigned the problem is the tun_srsue interface is not created

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
srsRan 23.3 using the gnb.yml and ue.conf config from the referenced guide

## Expected Behavior ##
[What you expect to happen]
The tun_srsue should get created in order for IP traffic flow to/from the simulated UE

## Actual Behaviour ##
[What happens instead e.g. error message]
The tun_srsue is not created, I run 'ip link' and it doesn't show. I can't ping either 

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
Compile srsran_4G and srsran with zmq support
Use the config files provided  and then run it with
$ sudo ip netns add ue1
$ sudo ./srsue ue.conf
Built in Release mode using commit 921f17484 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://gnb-zmq:2000,base_srate=11.52e6
Supported RF device list: UHD bladeRF zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://gnb-zmq:2000
CH0 tx_port=tcp://*:2001
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=174
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
PDU Session Establishment successful. IP: 172.16.10.14
RRC NR reconfiguration successful.

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

I'm running both gnb and srsue on Kubernetes. The pods have elevated privileged and tun interface can get created manually as I show below. I also tested the UERANSIM software UE and the tun interface gets properly created so I don't think kubernetes is the cause but it might be related.

$ sudo ip tuntap add mode tun dev tun0
$ ip link show de tun0
1: tun0: <POINTOPOINT,MULTICAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 500
    link/none

emunicio, 2023-05-09:
I may be experiencing the same issue.
However, while I see `tun_srsue` interface created with _ip netns exec ue1 ip addr_ , the UE can't ping the nor the GW neither its own IP.

herlesupreeth, 2023-05-11:
@gustavobsch As @emunicio mentioned, the tun interface gets created in the namespace you create (i.e. ue1). So in order to see the tun interface you need to execute `ip netns exec ue1 ip addr`

@emunicio Have you tried pinging as follows?

`ip netns exec ping <IP_ADDRESS>`

emunicio, 2023-05-25:
Yes, the problem was that the local IP of the conf file of the gnb was wrong ;-)
I guess we can close this

closed 2023-06-01 by ismagom