srsran/srsRAN_4G #1168: Connection refused when running 5G SA end to end

leovassalo, 2023-05-12:
## Issue Description ##
I am trying to follow the steps in the  [5G SA end to end tutorial](https://docs.srsran.com/projects/4g/en/latest/app_notes/source/5g_sa_E2E/source/index.html), but i get the following console response:

```
srsLog error - Unable to create log file "/tmp/enb.log": Permission denied
connect(): Connection refused
Failed to initiate NG connection. Attempting reconnection in 10 seconds
Opening 1 channels in RF device=zmq with args=fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=11.52e6
Supported RF device list: UHD bladeRF zmq file
CHx base_srate=11.52e6
CHx id=enb
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2000
CH0 fail_on_disconnect=true

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Setting frequency: DL=1842.5 Mhz, DL_SSB=1842.05 Mhz (SSB-ARFCN=368410), UL=1747.5 MHz for cc_idx=0 nof_prb=52
```
As you can see, the NG connection is being refused.
I believe I have strictly followed the instructions regarding the IP addresses mods.

Does allocating the ```epc.conf``` on the same folder as the ```srsenb``` script (which would be ```/srsRAN_4G/build/srsenb/src```, in my case) make any difference?

Otherwise, I cannot think of another reason for this error.

I can provide additional info if needed.
Thank you.

brendan-mcauliffe, 2023-05-19:
You will see from the tutorial that you **cannot** use srsEPC for 5G use-cases, we recommend using Open5GS as the 5G core. 

You need to make sure that in the srsENB config file that the MME address matches the GTPU address in the UPF and the NGAP address in the AMF of the Open5GS Core. Once these addresses are the same, and the PLMN and TAC being used are consistent between the Core and the srsENB configs then you should get a successful NG connection.

leovassalo, 2023-05-30:
Thank you, Brendan! The NG connection is now successful. Changing the NGAP and GTPU addresses solved the issue. However, I seem to have another problem.

when running the srsUE, I get the following feedback:

```Built in Release mode using commit 921f17484 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=11.52e6
Supported RF device list: UHD bladeRF zmq file
CHx base_srate=11.52e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=491
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
RRC NR reconfiguration successful.
Received RRC Release
```

As you can see, the PDU session is not properly established, and the device disconnects right after connecting.

The srsENB bash shows this:

```srsLog error - Unable to create log file "/tmp/enb.log": Permission denied
Opening 1 channels in RF device=zmq with args=fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=11.52e6
Supported RF device list: UHD bladeRF zmq file
CHx base_srate=11.52e6
CHx id=enb
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2000
CH0 fail_on_disconnect=true
NG connection successful

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Setting frequency: DL=1842.5 Mhz, DL_SSB=1842.05 Mhz (SSB-ARFCN=368410), UL=1747.5 MHz for cc_idx=0 nof_prb=52
RACH:  slot=491, cc=0, preamble=0, offset=0, temp_crnti=0x4601
Disconnecting rnti=0x4601.
```

In the Open5GS WebUI, the UE is configured with the same OPc, Key and IMSI values as the `ue.conf` file
Has this happened before? Do you have any idea on how I could solve this?

Thanks in advance!

brendan-mcauliffe, 2023-05-30:
I suggest you check how your APN is configured, you should make sure it is the same between srsUE and Open5GS. Typically if the PDU session isn't established it is because of a misconfiguration of the APN.

leovassalo, 2023-06-01:
Thank you, I was able to establish a PDU session now. The APN was misconfigured in the webUI. However, I still get the RRC release message shortly after connecting to the network.
Is this a common issue? Does it have to do with the IP configuration?

Thanks again!

brendan-mcauliffe, 2023-06-01:
Can you send logs and pcaps? Make sure to set your logs to "info". Several factors can cause an RRC Release to be sent, for example, channel conditions, poor sync between the UE and eNB, and issues arising from a lack of computational resources.

ismagom, 2023-06-01:
Please use srsRAN_Project for 5G SA

closed 2023-06-01 by ismagom