srsran/srsRAN_4G #1170: COTS UE cannot UL data communication for BW20MHz on 4G with RF board USRP

fj0831ec, 2023-05-17:
## Issue Description ##
[Describe the issue in detail]
I have verified COTS UE data communication over LTE using the combination of srsRAN 4G and USRP B200-mini shown below.
I have checked Bandwidth 20MHz.
COTS UE successfully LTE Attach and PDN Connect without any problem, but Reestablishment is repeated when UL data(U-plane) communication is started.
and.
[INFO][UHD RF] Tx while waiting for EOB, time out ...
is occurs frequently.
Radio will not be output from eNB and COTS UE will be out of service.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
Operation System Ubuntu 22.04.02 LTS
Hardware  Intel® Celeron(R) N5105 @ 2.00GHz × 4 with 8 GiB of Memory
RF front-end USRP B210-mini
library and driver versions  UHD_3.15.0.0-74-ge35f66e8

## Expected Behavior ##
[What you expect to happen]
Communication is possible without disconnection
## Actual Behaviour ##
[What happens instead e.g. error message]
$ sudo srsenb 
Active RF plugins: libsrsran_rf_uhd.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /root/.config/srsran/enb.conf
Couldn't open /root/.config/srsran/enb.conf either, trying /etc/srsran/enb.conf
Reading configuration file /etc/srsran/enb.conf...
Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
Couldn't open /root/.config/srsran/sib.conf either, trying /etc/srsran/sib.conf
Couldn't open rr.conf, trying /root/.config/srsran/rr.conf
Couldn't open /root/.config/srsran/rr.conf either, trying /etc/srsran/rr.conf
Couldn't open rb.conf, trying /root/.config/srsran/rb.conf
Couldn't open /root/.config/srsran/rb.conf either, trying /etc/srsran/rb.conf
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using commit 00c972ac4 on branch master.

Opening 1 channels in RF device=default with args=default
Supported RF device list: UHD file
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_3.15.0.0-74-ge35f66e8
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B200mini
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=2120.0 Mhz, UL=1930.0 MHz for cc_idx=0 nof_prb=100
RACH:  tti=3151, cc=0, pci=1, preamble=32, offset=1, temp_crnti=0x46
User 0x46 connected
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 25.1793 >= 25.1792. Starting new burst...
Disconnecting rnti=0x46.
RACH:  tti=5121, cc=0, pci=1, preamble=51, offset=1, temp_crnti=0x47
RRCReestablishmentReject for rnti=0x47. Cause: no context available
Disconnecting rnti=0x47.
RACH:  tti=5251, cc=0, pci=1, preamble=31, offset=1, temp_crnti=0x48
User 0x48 connected
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 28.0052 >= 25.1792. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 28.0141 >= 25.1792. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 28.0232 >= 25.1792. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 28.033 >= 25.1792. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 28.047 >= 25.1792. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 28.062 >= 25.1792. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 28.0734 >= 25.1792. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 30.0828 >= 30.0828. Starting new burst...
RACH:  tti=9061, cc=0, pci=1, preamble=47, offset=3, temp_crnti=0x49
User 0x49 requesting RRC Reestablishment as 0x48. Cause: otherFailure
Disconnecting rnti=0x48.
User 0x49 connected
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 34.2888 >= 34.2888. Starting new burst...
RACH:  tti=2941, cc=0, pci=1, preamble=5, offset=1, temp_crnti=0x4a
User 0x4a requesting RRC Reestablishment as 0x49. Cause: otherFailure
Disconnecting rnti=0x49.
User 0x4a connected
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 46.2616 >= 34.2888. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 46.2724 >= 34.2888. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 46.2842 >= 34.2888. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 46.2943 >= 34.2888. Starting new burst...
RACH:  tti=3601, cc=0, pci=1, preamble=14, offset=1, temp_crnti=0x4b
User 0x4b requesting RRC Reestablishment as 0x4a. Cause: otherFailure
Disconnecting rnti=0x4a.
User 0x4b connected
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 53.49 >= 48.4071. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 53.4988 >= 48.4071. Starting new burst...
RACH:  tti=531, cc=0, pci=1, preamble=44, offset=1, temp_crnti=0x4c
User 0x4c requesting RRC Reestablishment as 0x4b. Cause: otherFailure
Disconnecting rnti=0x4b.
User 0x4c connected
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 56.2741 >= 55.6112. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 56.286 >= 55.6112. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 56.2954 >= 55.6112. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 56.3049 >= 55.6112. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 58.3113 >= 58.3113. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 59.2463 >= 58.3113. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 59.2625 >= 58.3113. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 59.2645 >= 58.3113. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 59.2784 >= 58.3113. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 59.2878 >= 58.3113. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 59.2977 >= 58.3113. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 59.3072 >= 58.3113. Starting new burst...
Disconnecting rnti=0x4c.
^CStopping ..
---  exiting  ---

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
Change the following settings in the enb.conf 
n_prb = 100
dl_earfcn = 100

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
No problem with Bandwidth=10MHz

nikitos1550, 2023-05-28:
In enb.conf you can find following
```
# For best performance in 2x2 MIMO and >= 15 MHz use the following device_args settings:
#     USRP B210: num_recv_frames=64,num_send_frames=64
#     And for 75 PRBs, also append ",master_clock_rate=15.36e6" to the device args
```
You can try add device_args with `num_recv_frames=64,num_send_frames=64`.
The issue you get seems something about performance. You can try same configuration on more powerful PC and compare results.

fj0831ec, 2023-05-29:
Thank you for answering.
So, when I changed the specs of my PC, the problem no longer occurred.
Changed PC specs：
Intel(R) Core(TM) i5-8279U CPU @ 2.40GHz with 6 GiB of Memory

closed 2023-05-29 by fj0831ec