srsran/srsRAN_4G #1171: need some help with connecting phone to srsran 4G

nikitos1550, 2023-05-21:
## Issue Description ##
I setuped srsran (release 23.04) on ubuntu 22.04 with b210.
Everything build fine and I can start software withoyt any issue.
I programmed custom sim card (chinese green card) and added it's info to db.
My android phone (I used two different) can see the network and try connect to it, but phone cannot register the network for some reason. 
Here is log, for some reason it start attach procedure again and again from the beginning.

```
2023-05-21T13:33:34.733925 [S1AP   ] [D] Saved UE context corresponding to IMSI 901550000123456
2023-05-21T13:33:34.733925 [S1AP   ] [D] Saved UE context corresponding to MME UE S1AP Id 6
2023-05-21T13:33:34.733926 [S1AP   ] [D] Added UE with MME-UE S1AP Id 6 to eNB with association 212
2023-05-21T13:33:34.733927 [NAS    ] [I] Packing Authentication Request
2023-05-21T13:33:34.733928 [S1AP   ] [D] Sending message to eNB with SCTP association 212. MME UE S1AP ID 6, eNB UE S1AP ID 1
2023-05-21T13:33:34.733933 [S1AP   ] [D] Transmitting S1AP PDU. eNB SCTP association Id: 212
2023-05-21T13:33:34.733966 [NAS    ] [I] Downlink NAS: Sending Authentication Request
2023-05-21T13:33:34^[[6~.733970 [S1AP   ] [D] Waiting for S1-MME or S11 Message
2023-05-21T13:33:59.775983 [S1AP   ] [I] Received S1AP msg. Size: 116
2023-05-21T13:33:59.775998 [S1AP   ] [I] Received Initiating PDU
2023-05-21T13:33:59.775999 [S1AP   ] [I] Received Initial UE Message.
2023-05-21T13:33:59.776012 [S1AP   ] [I] Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
2023-05-21T13:33:59.776018 [S1AP   ] [I] Received Initial UE message -- Attach Request
2023-05-21T13:33:59.776026 [NAS    ] [I] Attach request -- IMSI: 901550000123456
2023-05-21T13:33:59.776029 [NAS    ] [I] Attach request -- eNB-UE S1AP Id: 2
2023-05-21T13:33:59.776031 [NAS    ] [I] Attach request -- Attach type: 2
2023-05-21T13:33:59.776034 [NAS    ] [I] Attach Request -- UE Network Capabilities EEA: 11110000
2023-05-21T13:33:59.776036 [NAS    ] [I] Attach Request -- UE Network Capabilities EIA: 01110000
2023-05-21T13:33:59.776039 [NAS    ] [I] Attach Request -- MS Network Capabilities Present: true
2023-05-21T13:33:59.776042 [NAS    ] [I] PDN Connectivity Request -- EPS Bearer Identity requested: 0
2023-05-21T13:33:59.776046 [NAS    ] [I] PDN Connectivity Request -- Procedure Transaction Id: 121
2023-05-21T13:33:59.776049 [NAS    ] [I] PDN Connectivity Request -- ESM Information Transfer requested: true
2023-05-21T13:33:59.776050 [NAS    ] [I] Attach Request -- Found previously attached UE.
2023-05-21T13:33:59.776053 [MME GTPC] [I] Sending GTP-C Delete Session Request request. IMSI 901550000123456
2023-05-21T13:33:59.776054 [MME GTPC] [E] Could not find GTP-C context to remove
2023-05-21T13:33:59.776055 [S1AP   ] [E] UE is not ECM connected. No send context release command. MME UE S1AP Id 6
2023-05-21T13:33:59.776058 [S1AP   ] [I] Released UE ECM Context.
2023-05-21T13:33:59.776059 [S1AP   ] [I] Deleted UE Context.
2023-05-21T13:33:59.776061 [NAS    ] [D] NAS Context Initialized. MCC: 0xf901, MNC 0xff55
2023-05-21T13:33:59.776062 [HSS    ] [D] Generating AUTH info answer
2023-05-21T13:33:59.776065 [HSS    ] [D] User Key : 
    0000: f4 7f 37 e4 71 9a 9c 37 9e 34 47 f0 89 b1 f1 0a
2023-05-21T13:33:59.776066 [HSS    ] [D] User OPc : 
    0000: f4 7f 37 e4 71 9a 9c 37 9e 34 47 f0 89 b1 f1 0b
2023-05-21T13:33:59.776066 [HSS    ] [D] User Rand : 
    0000: f7 d7 58 ae d8 e6 6a 0b ac ec 3c f0 b1 57 0c 0e
2023-05-21T13:33:59.776067 [HSS    ] [D] User XRES: 
    0000: ef 87 83 34 77 70 a7 bd
2023-05-21T13:33:59.776067 [HSS    ] [D] User CK: 
    0000: 94 fb 68 81 1f ee 59 d3 74 f3 3a ac e8 2c 75 fb
2023-05-21T13:33:59.776067 [HSS    ] [D] User IK: 
    0000: 12 7d 95 5d 0a bb 27 b4 a6 d4 8e f6 74 5a 1a 01
2023-05-21T13:33:59.776068 [HSS    ] [D] User AK: 
    0000: 9f 4b 68 49 09 48
2023-05-21T13:33:59.776068 [HSS    ] [D] User SQN : 
    0000: 00 00 00 00 21 0f
2023-05-21T13:33:59.776069 [HSS    ] [D] User MAC : 
    0000: 5d 5b 1f 78 5b 4a 23 6b
2023-05-21T13:33:59.776075 [HSS    ] [D] User MCC : f901  MNC : ff55 
2023-05-21T13:33:59.776075 [HSS    ] [D] User k_asme : 
    0000: fe 20 c8 a9 26 a0 2e 77 9d e3 e9 2c ee 0e 42 ed
    0010: 68 c2 f6 8f 72 99 32 3f b5 47 29 1a a7 12 83 1d
2023-05-21T13:33:59.776075 [HSS    ] [D] User AUTN: 
    0000: 9f 4b 68 49 28 47 70 00 5d 5b 1f 78 5b 4a 23 6b
2023-05-21T13:33:59.776075 [HSS    ] [D] Incremented SQN  -- IMSI: 901550000123456
2023-05-21T13:33:59.776076 [HSS    ] [D] SQN: 
    0000: 00 00 00 00 21 30
2023-05-21T13:33:59.776077 [S1AP   ] [D] Saved UE context corresponding to IMSI 901550000123456
2023-05-21T13:33:59.776077 [S1AP   ] [D] Saved UE context corresponding to MME UE S1AP Id 7
2023-05-21T13:33:59.776077 [S1AP   ] [D] Added UE with MME-UE S1AP Id 7 to eNB with association 212
2023-05-21T13:33:59.776078 [NAS    ] [I] Packing Authentication Request
2023-05-21T13:33:59.776079 [S1AP   ] [D] Sending message to eNB with SCTP association 212. MME UE S1AP ID 7, eNB UE S1AP ID 2
[... 328 lines cut ...]
lte    1   4a    2   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
lte    1   49  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
lte    1   4a    1   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0

               -----------------DL----------------|-------------------------UL-------------------------
rat  pci rnti  cqi  ri  mcs  brate   ok  nok  (%) | pusch  pucch  phr  mcs  brate   ok  nok  (%)    bsr
lte    1   49  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
lte    1   4a    0   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
lte    1   49  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
lte    1   4a    0   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
lte    1   49  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
lte    1   4a    2   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
lte    1   49  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
lte    1   4a    2   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
lte    1   49  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
lte    1   4a    0   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
lte    1   49  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
lte    1   4a    0   0    0      0    0    0   0% |   n/a   99.9    0    0      0    0    0   0%    0.0
Disconnecting rnti=0x49.
lte    1   4a    5   0    0      0    0    0   0% |   n/a    3.7    0    0      0    0    0   0%    0.0
lte    1   4a  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
lte    1   4a  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
lte    1   4a  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
lte    1   4a  n/a   0    0      0    0    0   0% |   n/a    n/a    0    0      0    0    0   0%    0.0
```

nikitos1550, 2023-05-21:
Here is another log
```

Built in Release mode using 23.04.0.

---  Software Radio Systems EPC  ---

Couldn't open , trying /root/.config/srsran/epc.conf
Reading configuration file /root/.config/srsran/epc.conf...
Couldn't open user_db.csv, trying /root/.config/srsran/user_db.csv
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf901, MNC: 0xff55
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:901, MNC:55
S1 Setup Request - TAC 7, B-PLMN 0x9f155
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI: 901550000123456
Attach request -- eNB-UE S1AP Id: 1
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 115
PDN Connectivity Request -- ESM Information Transfer requested: true
Downlink NAS: Sending Authentication Request
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI: 901550000123456
Attach request -- eNB-UE S1AP Id: 2
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 116
PDN Connectivity Request -- ESM Information Transfer requested: true
Attach Request -- Found previously attach UE.
Downlink NAS: Sending Authentication Request
Received UE Context Release Request. MME-UE S1AP Id 1
No UE context to release found. MME-UE S1AP Id: 1
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI: 901550000123456
Attach request -- eNB-UE S1AP Id: 3
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 117
PDN Connectivity Request -- ESM Information Transfer requested: true
Attach Request -- Found previously attach UE.
Downlink NAS: Sending Authentication Request
Received UE Context Release Request. MME-UE S1AP Id 2
No UE context to release found. MME-UE S1AP Id: 2
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI: 901550000123456
Attach request -- eNB-UE S1AP Id: 4
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 118
PDN Connectivity Request -- ESM Information Transfer requested: true
Attach Request -- Found previously attach UE.
Downlink NAS: Sending Authentication Request
Received UE Context Release Request. MME-UE S1AP Id 3
No UE context to release found. MME-UE S1AP Id: 3
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI: 901550000123456
Attach request -- eNB-UE S1AP Id: 5
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 119
PDN Connectivity Request -- ESM Information Transfer requested: true
Attach Request -- Found previously attach UE.
Downlink NAS: Sending Authentication Request
Received UE Context Release Request. MME-UE S1AP Id 4
```

andrepuschmann, 2023-05-22:
You have to check your RF settings. The CQI reported by the phone is very low and also the UL SNR isn't good.

nikitos1550, 2023-05-22:
Thanks, will try find and replace antennas with better. 
Also I recently got the connection with phone, with another sim card sample (mostly same settings except imsi). So maybe you can tell me some rules, what sim card settings are really important? Also are there any restrictions on Ki, OPC values (like it cannot be all 00 or all FF etc)? 
I will make more experiment and post results here later.

nikitos1550, 2023-05-22:
P.S. Here https://pasteboard.co/ZRlftMIqE6o0.jpg is screenshot of success smartphone connection, is CQI also low?

andrepuschmann, 2023-05-22:
This time the cqi is good. Max value is 15. Seems those RF settings work better.

closed 2023-06-01 by ismagom