srsran/srsRAN_4G #1173: How to decode npdsch.pcap?

VLiu7, 2023-05-23:
## Issue Description ##
Hi!
I followed this tutorial (https://docs.srsran.com/projects/4g/en/latest/app_notes/source/nbiot/source/index.html) to set up a NB-IoT testbed. 
But I cannot decode `npdsch.pcap` with wireshark.
I found this pcap is with `DLT` 147, but this tutorial(https://docs.srsran.com/projects/4g/en/latest/usermanuals/source/srsue/source/2_ue_getstarted.html?highlight=pcap#observing-results) didn't cover 147. 
How should I config my wireshark to decode the pcap?
This is an example [pcap](https://github.com/srsran/srsRAN_4G/files/11541094/npdsch.pcap.zip).
And I confirmed that it worked well with srsGUI.

[image]

Thanks in advance!

brendan-mcauliffe, 2023-06-09:
Hi @VLiu7, 

If you go _Preferences > Protocols > DLT_USER_

You can add an entry with the following details for `DLT=147` to correctly decode the pcap: 

[image]

(See the highlighted line)

VLiu7, 2023-06-11:
It works! Thanks a lot!

closed 2023-06-11 by VLiu7