srsran/srsRAN_4G #1174: [Advice needed] Is it possible to make srsRAN accept any APN or embedd apn name to simcard?

nikitos1550, 2023-05-23:
How usually smartphone knows what APN use for particular sim card (operator)?
Now I am adding apn manually to my smartphone, but better to let it automatically use some APN.
Also maybe it is possible let sraRAN accept connections with any apn.

ismagom, 2023-06-01:
This is set in the MME configuration

closed 2023-06-01 by ismagom

nathantsmith, 2023-12-20:
not sure that this is  answers the original reporter's question.  Some devices may have fixed set of APN(s) that cannot be changed. In my case, I'd like to setup multiple APNs  and point them all to the same gateway. Or make some modification to SRSRAN to just accept any APN request and forward everything to the same gateway.   Is it possible to have SRSRAN accept any APN or configure multiple APN's specifically? @ismagom