srsran/srsRAN_4G #1175: GPSDO for base station case

nikitos1550, 2023-05-25:
Hello!

Can GPSDO improve something in case with one usrp b210 LTE base station case?

matan1008, 2023-05-25:
From my experience the answer is yes, and dramatically

andrepuschmann, 2023-05-25:
Hey @nikitos1550. Please use the discussion feature https://github.com/srsran/srsRAN_4G/discussions for questions like this. It's not a code issue and therefore shouldn't be discussed in the issues section.

However, I can second @matan1008 response. Any external clock source that has a better frequency stability will improve the overall RF performance. It's just that the B210 or other cheap COTS SDR boards are not equipped with good enough oscillators  and they usually vary a lot between badges.

closed 2023-05-25 by andrepuschmann