srsran/srsRAN_4G #1185: UE security capabilities mismatch

grmono, 2023-06-11:
## Issue Description ##
I am trying to connect a Samsung A130 to the 4G srsenb/srsepc setup but fail to complete the Security Command flow due to UE capabilities mismatch send from the UE side

## Setup Details ##
Ettus B200 as SRSENB and samsung A130 as UE with valid USIM, i am using the latest version of srsRAN_4G , on both srsepc and srsenb i am uzing EIA =0 and EEA = 0.

## Expected Behavior ##
Security capabilities should match the capabilities received by the UE and the UE should not send security command reject due to ue capabilties

## Actual Behaviour ##
The UE replied with EMM cause Cause: UE security capabilities mismatch (23)

Their is an issue found on open5gs that might be related to this issue here
https://github.com/open5gs/open5gs/issues/27

**Security Capabilities send by UE on attach request**

UE network capability
    Length: 7

    1... .... = EEA0: Supported
    .1.. .... = 128-EEA1: Supported
    ..1. .... = 128-EEA2: Supported
    ...1 .... = 128-EEA3: Supported
    .... 0... = EEA4: Not supported
    .... .0.. = EEA5: Not supported
    .... ..0. = EEA6: Not supported
    .... ...0 = EEA7: Not supported
    1... .... = EIA0: Supported
    .1.. .... = 128-EIA1: Supported
    ..1. .... = 128-EIA2: Supported
    ...1 .... = 128-EIA3: Supported
    .... 0... = EIA4: Not supported
    .... .0.. = EIA5: Not supported
    .... ..0. = EIA6: Not supported
    .... ...0 = EIA7: Not supported
    1... .... = UEA0: Supported
    .1.. .... = UEA1: Supported
    ..0. .... = UEA2: Not supported
    ...0 .... = UEA3: Not supported
    .... 0... = UEA4: Not supported
    .... .0.. = UEA5: Not supported
    .... ..0. = UEA6: Not supported
    .... ...0 = UEA7: Not supported
    1... .... = UCS2 support (UCS2): The UE has no preference between the use of the default alphabet and the use of UCS2
    .1.. .... = UMTS integrity algorithm UIA1: Supported
    ..0. .... = UMTS integrity algorithm UIA2: Not supported
    ...0 .... = UMTS integrity algorithm UIA3: Not supported
    .... 0... = UMTS integrity algorithm UIA4: Not supported
    .... .0.. = UMTS integrity algorithm UIA5: Not supported
    .... ..0. = UMTS integrity algorithm UIA6: Not supported
    .... ...0 = UMTS integrity algorithm UIA7: Not supported
    0... .... = ProSe direct discovery: Not supported
    .0.. .... = ProSe: Not supported
    ..0. .... = H.245 After SRVCC Handover: Not supported
    ...1 .... = Access class control for CSFB: Supported
    .... 0... = LTE Positioning Protocol: Not supported
    .... .0.. = Location services (LCS) notification mechanisms: Not supported
    .... ..0. = SRVCC from E-UTRAN to cdma2000 1xCS: Not supported
    .... ...1 = Notification procedure: Supported
    0... .... = Extended protocol configuration options: Not supported
    .0.. .... = Header compression for control plane CIoT EPS optimization: Not supported
    ..0. .... = EMM-REGISTERED w/o PDN connectivity: Not supported
    ...0 .... = S1-U data transfer: Not supported
    .... 0... = User plane CIoT EPS optimization: Not supported
    .... .0.. = Control plane CIoT EPS optimization: Not supported
    .... ..0. = ProSe UE-to-network relay: Not supported
    .... ...0 = ProSe direct communication: Not supported
    0... .... = Signalling for a maximum number of 15 EPS bearer contexts: Not supported
    .0.. .... = Service gap control: Not supported
    ..0. .... = N1 mode: Not supported
    ...0 .... = Dual connectivity with NR: Not supported
    .... 0... = Control plane data backoff: Not supported
    .... .0.. = Restriction on use of enhanced coverage: Not supported
    .... ..0. = V2X communication over PC5: Not supported
    .... ...0 = Multiple DRB: Not supported

**Capabilities Replayed in security command**

UE security capability - Replayed UE security capabilities
    Length: 5

    1... .... = EEA0: Supported
    .1.. .... = 128-EEA1: Supported
    ..1. .... = 128-EEA2: Supported
    ...1 .... = 128-EEA3: Supported
    .... 0... = EEA4: Not supported
    .... .0.. = EEA5: Not supported
    .... ..0. = EEA6: Not supported
    .... ...0 = EEA7: Not supported
    1... .... = EIA0: Supported
    .1.. .... = 128-EIA1: Supported
    ..1. .... = 128-EIA2: Supported
    ...1 .... = 128-EIA3: Supported
    .... 0... = EIA4: Not supported
    .... .0.. = EIA5: Not supported
    .... ..0. = EIA6: Not supported
    .... ...0 = EIA7: Not supported
    1... .... = UEA0: Supported
    .1.. .... = UEA1: Supported
    ..0. .... = UEA2: Not supported
    ...0 .... = UEA3: Not supported
    .... 0... = UEA4: Not supported
    .... .0.. = UEA5: Not supported
    .... ..0. = UEA6: Not supported
    .... ...0 = UEA7: Not supported
    0... .... = Spare bit(s): 0x0
    .1.. .... = UMTS integrity algorithm UIA1: Supported
    ..0. .... = UMTS integrity algorithm UIA2: Not supported
    ...0 .... = UMTS integrity algorithm UIA3: Not supported
    .... 0... = UMTS integrity algorithm UIA4: Not supported
    .... .0.. = UMTS integrity algorithm UIA5: Not supported
    .... ..0. = UMTS integrity algorithm UIA6: Not supported
    .... ...0 = UMTS integrity algorithm UIA7: Not supported
    0... .... = Spare bit(s): 0x0
    .0.. .... = GPRS encryption algorithm GEA1: Not supported
    ..1. .... = GPRS encryption algorithm GEA2: Supported
    ...1 .... = GPRS encryption algorithm GEA3: Supported
    .... 1... = GPRS encryption algorithm GEA4: Supported
    .... .0.. = GPRS encryption algorithm GEA5: Not supported
    .... ..0. = GPRS encryption algorithm GEA6: Not supported
    .... ...0 = GPRS encryption algorithm GEA7: Not supported

Tonyhauatust, 2024-06-11:
same issue encountered by me, any updates?

Tonyhauatust, 2024-06-16:
Updates: After disabling videotelephony functionality on my android phone the security command mode is successful.