srsran/srsRAN_4G #1186: Enb is not connected with UE by using Adalm Pluto,Error observed when srate=11.52e6

sruthi81, 2023-06-16:
sysnetlabcomp6@sysnetlabcomp6-HP-EliteDesk-800-G1-TWR:~/srsRAN_4G/build$ sudo ./srsue/src/srsue
Active RF plugins: libsrsran_rf_soapy.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Couldn't open , trying /root/.config/srsran/ue.conf
Reading configuration file /root/.config/srsran/ue.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using commit 5a4eb5d6c on branch master.

Opening 1 channels in RF device=soapy with args=default
Supported RF device list: soapy zmq file
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.4.0.0-0ubuntu1~jammy1
ERROR: Unable to claim interface 3:6:5: Device or resource busy (16)
Soapy has found device #0: device=PlutoSDR, driver=plutosdr, label=PlutoSDR #0 ip:pluto.local, uri=ip:pluto.local, 
Selecting Soapy device: 0
[INFO] Opening PlutoSDR #0 ip:pluto.local...
Setting up Rx stream with 1 channel(s)
[INFO] Using format CF32.
[INFO] Auto setting Buffer Size: 262144
[INFO] Set MTU Size: 262144
Setting up Tx stream with 1 channel(s)
[INFO] Using format CF32.
[INFO] Has direct TX copy: 1
[INFO] Auto setting Buffer Size: 32768
[INFO] Set MTU Size: 32768
Available device sensors: 
 - xadc_temp0
 - xadc_voltage0
 - xadc_voltage1
 - xadc_voltage2
 - xadc_voltage3
 - xadc_voltage4
 - xadc_voltage5
 - xadc_voltage6
 - xadc_voltage7
 - xadc_voltage8
 - adm1177_current0
 - adm1177_voltage0
 - ad9361-phy_temp0
 - ad9361-phy_voltage2
Available sensors for Rx channel 0: 
State of gain elements for Rx channel 0 (AGC supported):
 - PGA: 71.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PGA: 80.00 dB
Rx antenna set to A_BALANCED
Tx antenna set to A
[INFO] Auto setting Buffer Size: 262144
[INFO] Set MTU Size: 262144
/home/sysnetlabcomp6/srsRAN_4G/lib/src/radio/radio.cc:750: Assertion Failure: The sampling rate ratio is not integer (11.52 MHz / 11.52 MHz = 1.000)
--- command='./srsue/src/srsue' version=22.10.0 signal=6 date='14/06/2023 11:50:23' ---
	./srsue/src/srsue(+0x319a0a) [0x55b15ef22a0a]
	/lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f127ea42520]
	/lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c) [0x7f127ea96a7c]
	/lib/x86_64-linux-gnu/libc.so.6(raise+0x16) [0x7f127ea42476]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0xd3) [0x7f127ea287f3]
	./srsue/src/srsue(+0x12a986) [0x55b15ed33986]
	./srsue/src/srsue(+0x472e7e) [0x55b15f07be7e]
	./srsue/src/srsue(+0x108816) [0x55b15ed11816]
	./srsue/src/srsue(+0xea839) [0x55b15ecf3839]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xdc2b3) [0x7f127eedc2b3]
	/lib/x86_64-linux-gnu/libc.so.6(+0x94b43) [0x7f127ea94b43]
	/lib/x86_64-linux-gnu/libc.so.6(+0x126a00) [0x7f127eb26a00]
srsRAN crashed. Please send this backtrace to the developers ...
---  exiting  ---

andrepuschmann, 2023-06-16:
That should be fixed with the current release thanks to https://github.com/srsran/srsRAN_4G/pull/1082

closed 2023-06-16 by andrepuschmann

sruthi81, 2023-06-24:
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_soapy.so libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /root/.config/srsran/enb.conf
Reading configuration file /root/.config/srsran/enb.conf...
Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
Couldn't open rr.conf, trying /root/.config/srsran/rr.conf
Couldn't open rb.conf, trying /root/.config/srsran/rb.conf
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using commit 5a4eb5d6c on branch master.

Opening 1 channels in RF device=soapy with args=default
Supported RF device list: UHD soapy zmq file
NG connection successful
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.4.0.0-0ubuntu1~jammy1
Soapy has found device #0: device=PlutoSDR, driver=plutosdr, label=PlutoSDR #0 usb:1.7.5, uri=usb:1.7.5, 
Soapy has found device #1: device=PlutoSDR, driver=plutosdr, label=PlutoSDR #0 ip:pluto.local, uri=ip:pluto.local, 
Selecting Soapy device: 0
[INFO] Opening PlutoSDR #0 usb:1.7.5...
Setting up Rx stream with 1 channel(s)
[INFO] Using format CF32.
[INFO] Auto setting Buffer Size: 262144
[INFO] Set MTU Size: 262144
Setting up Tx stream with 1 channel(s)
[INFO] Using format CF32.
[INFO] Has direct TX copy: 1
[INFO] Auto setting Buffer Size: 32768
[INFO] Set MTU Size: 32768
Available device sensors: 
 - xadc_temp0
 - xadc_voltage0
 - xadc_voltage1
 - xadc_voltage2
 - xadc_voltage3
 - xadc_voltage4
 - xadc_voltage5
 - xadc_voltage6
 - xadc_voltage7
 - xadc_voltage8
 - adm1177_current0
 - adm1177_voltage0
 - ad9361-phy_temp0
 - ad9361-phy_voltage2
Available sensors for Rx channel 0: 
State of gain elements for Rx channel 0 (AGC supported):
 - PGA: 20.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PGA: 89.00 dB
Rx antenna set to A_BALANCED
Tx antenna set to A

==== eNodeB started ===
Type <t> to view trace
[INFO] Auto setting Buffer Size: 262144
[INFO] Set MTU Size: 262144
Setting frequency: DL=1842.5 Mhz, DL_SSB=1842.05 Mhz (SSB-ARFCN=368410), UL=1747.5 MHz for cc_idx=0 nof_prb=52
[INFO] Has direct RX copy: 1

> Sir, new release has updated, but still getting this error. 
#####################################################################
#                   srsENB configuration file
#####################################################################

#####################################################################
# eNB configuration
#
# enb_id:               20-bit eNB identifier.
# mcc:                  Mobile Country Code
[... 188 lines cut ...]
hostname = pluto
ipaddr = 192.168.2.1
ipaddr_host = 192.168.2.10
netmask = 255.255.255.0

[WLAN]
ssid_wlan = 
pwd_wlan = 
ipaddr_wlan = 

[USB_ETHERNET]
ipaddr_eth = 
netmask_eth = 255.255.255.0

[SYSTEM]
xo_correction = 
udc_handle_suspend = 0
# USB Communication Device Class Compatibility Mode [rndis|ncm|ecm]
usb_ethernet_mode = 0

[ACTIONS]
diagnostic_report = 1
dfu = 0
reset = 0
calibrate = 0

sruthi81, 2023-07-18:
> That should be fixed with the current release thanks to #1082

I am sorry, the assertion error was gone when am using 15.36. but ue is working with 15.36 and enb is working with 11.52e6 , so there is no progress in communication