srsran/srsRAN_4G #1188: srsUE not starting while using ADALM Pluto hardware device

Bipro002, 2023-06-23:
## Issue Description ##
Getting the error message "srsRAN crashed. Please send this backtrace to the developers ..." while starting srsUE

## Setup Details ##
We are trying to build a 5G network between 3 computers using srsRAN, Open5gs and 2 ADALM Pluto devices. On one computer we are running the core network using Open5gs. Another adjacent computer is running srsENB. These two computers are connected using ethernet cable. On the 3rd computer we are running srsUE. We are using 2 ADALM Pluto devices to communicate between srsENB and srsUE. 
## Expected Behavior ##
srsUE connects to srsENB and ping can be sent between the computers.

## Actual Behaviour ##

Error observed while running srsUE:

Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_soapy.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Couldn't open , trying /root/.config/srsran/ue.conf
Reading configuration file /root/.config/srsran/ue.conf...

Built in Release mode using commit fa56836b1 on branch master.

srsLog error - Unable to create log file "/tmp/ue.log": Permission denied
Opening 1 channels in RF device=soapy with args=default
Supported RF device list: UHD soapy zmq file
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.4.0.0-0ubuntu1~jammy1
Soapy has found device #0: device=PlutoSDR, driver=plutosdr, label=PlutoSDR #0 usb:3.7.5, uri=usb:3.7.5, 
Soapy has found device #1: device=PlutoSDR, driver=plutosdr, label=PlutoSDR #0 ip:pluto.local, uri=ip:pluto.local, 
Selecting Soapy device: 0
[INFO] Opening PlutoSDR #0 usb:3.7.5...
Setting up Rx stream with 1 channel(s)
[INFO] Using format CF32.
[INFO] Auto setting Buffer Size: 262144
[INFO] Set MTU Size: 262144
Setting up Tx stream with 1 channel(s)
[INFO] Using format CF32.
[INFO] Has direct TX copy: 1
[INFO] Auto setting Buffer Size: 32768
[INFO] Set MTU Size: 32768
Available device sensors: 
 - xadc_temp0
 - xadc_voltage0
 - xadc_voltage1
 - xadc_voltage2
 - xadc_voltage3
 - xadc_voltage4
 - xadc_voltage5
 - xadc_voltage6
 - xadc_voltage7
 - xadc_voltage8
 - adm1177_current0
 - adm1177_voltage0
 - ad9361-phy_temp0
 - ad9361-phy_voltage2
Available sensors for Rx channel 0: 
State of gain elements for Rx channel 0 (AGC supported):
 - PGA: 71.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PGA: 80.00 dB
Rx antenna set to A_BALANCED
Tx antenna set to A
Waiting PHY to initialize ... [INFO] Auto setting Buffer Size: 262144
[INFO] Set MTU Size: 262144
/home/sysnetlabcomp6/srsRAN_4G/lib/src/radio/radio.cc:750: Assertion Failure: The sampling rate ratio is not integer (11.52 MHz / 11.52 MHz = 1.000)
--- command='./srsue/src/srsue' version=23.04.0 signal=6 date='22/06/2023 07:36:55' ---
	./srsue/src/srsue(+0x32cd5a) [0x55b54ad98d5a]
	/lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f5fc1042520]
	/lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c) [0x7f5fc1096a7c]
	/lib/x86_64-linux-gnu/libc.so.6(raise+0x16) [0x7f5fc1042476]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0xd3) [0x7f5fc10287f3]
	./srsue/src/srsue(+0x12f416) [0x55b54ab9b416]
	./srsue/src/srsue(+0x48624e) [0x55b54aef224e]
	./srsue/src/srsue(+0x10d286) [0x55b54ab79286]
	./srsue/src/srsue(+0xef2a9) [0x55b54ab5b2a9]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xdc2b3) [0x7f5fc14dc2b3]
	/lib/x86_64-linux-gnu/libc.so.6(+0x94b43) [0x7f5fc1094b43]
	/lib/x86_64-linux-gnu/libc.so.6(+0x126a00) [0x7f5fc1126a00]
srsRAN crashed. Please send this backtrace to the developers ...
---  exiting  ---

## Steps to reproduce the problem ##

## Additional Information ##

Bipro002, 2023-07-07:
The error was resolved when we used srate=15.36MHz in ue.conf but UE cannot attach to eNB. Any suggestions would be really helpful
Thanks,
Bipro

vol4iniche, 2025-01-19:
Hello @Bipro002 , did you resolve the issue? 
I have the same ;(