srsran/srsRAN_4G #1192: dmrs_add_position option not present with srsUE and latest gnb from srsran_Project

koh-osug, 2023-07-01:
## Issue Description ##

the srsUE  cannot connect anymore to the Nb binary since the latest update to the master of srsran_Project. I get:

~~~
2023-07-01T21:16:37.204196 [ASN1   ] [W] [    0] dmrs_add_position option not present
2023-07-01T21:16:37.204199 [ASN1   ] [E] [    0] The enum of type dmrs_ul_cfg_s::dmrs_add_position_e_ was not initialized.
2023-07-01T21:16:37.204200 [ASN1   ] [W] [    0] Invalid option for dmrs_add_position 
2023-07-01T21:16:37.204200 [RRC-NR ] [W] Warning while build srsran_dmrs_sch_add_pos structure
~~~

## Setup Details ##

I'm using ZMQ with free5gc and had no issues before the update.

herlesupreeth, 2023-07-02:
This has been fixed in latest commit please pull the latest changes and retry

koh-osug, 2023-07-02:
Great. This helps.

closed 2023-07-02 by koh-osug