srsran/srsRAN_4G #1193: srsue can't decode paging msg

hz-nokia, 2023-07-03:
Hello SRSRAN team,

Recently I used B200 to test srsue,  I found paging decode failure in ue.log as following:

2023-07-03T01:38:30.475924 [RRC    ] [I] Proc "Process PCCH" - starting...
2023-07-03T01:38:30.475925 [RRC    ] [I] Proc "Process PCCH" - Received paging (1/1) for UE 180:18446744072942733775
2023-07-03T01:38:30.475925 [RRC    ] [I] Proc "Process PCCH" - Received paging for unknown identity
2023-07-03T01:38:30.475925 [RRC    ] [I] Proc "Process PCCH" - Completed successfully
2023-07-03T01:38:30.486915 [RRC    ] [I] Proc "Cell Reselection" - Starting...
2023-07-03T01:38:30.486916 [RRC    ] [D] Proc "Cell Selection" - Skipping cell selection procedure as there are no stronger neighbours in same EARFCN.
2023-07-03T01:38:30.486917 [RRC    ] [I] Proc "Cell Selection" - Completed with success.
2023-07-03T01:38:30.486917 [RRC    ] [I] Proc "Cell Reselection" - Cell Selection completed. Handling its result...
2023-07-03T01:38:30.486918 [RRC    ] [I] Proc "Cell Reselection" - Finished successfully
2023-07-03T01:38:30.495857 [RRC    ] [D] MEAS:  Processing measurement of 1 cells
2023-07-03T01:38:30.495857 [RRC    ] [I] MEAS:  New measurement serving cell: earfcn=0, pci=264, rsrp=-75.36 dBm, cfo=+2879.1 Hz
2023-07-03T01:38:30.505099 [RLC    ] [I] PCCH message received.
    0000: 48 1b 22 2a fc 41 83
2023-07-03T01:38:30.505928 [ASN1   ] [E] [    0] The condition lb <= n <= ub (0 <= 10 <= 9) was not met
2023-07-03T01:38:30.505929 [ASN1   ] [E] [    0] [/home/dz/srsRAN/lib/src/asn1/asn1_utils.cc][890] Decoding failure.
2023-07-03T01:38:30.505929 [ASN1   ] [E] [    0] [/home/dz/srsRAN/lib/include/srsran/asn1/asn1_utils.h][1079] Decoding failure.
2023-07-03T01:38:30.505930 [ASN1   ] [E] [    0] [/home/dz/srsRAN/lib/src/asn1/rrc/paging.cc][481] Decoding failure.
2023-07-03T01:38:30.505930 [ASN1   ] [E] [    0] [/home/dz/srsRAN/lib/src/asn1/rrc/paging.cc][546] Decoding failure.
2023-07-03T01:38:30.505930 [ASN1   ] [E] [    0] [/home/dz/srsRAN/lib/include/srsran/asn1/asn1_utils.h][1091] Decoding failure.
2023-07-03T01:38:30.505930 [ASN1   ] [E] [    0] [/home/dz/srsRAN/lib/src/asn1/rrc/paging.cc][633] Decoding failure.
2023-07-03T01:38:30.505930 [ASN1   ] [E] [    0] [/home/dz/srsRAN/lib/src/asn1/rrc/paging.cc][740] Decoding failure.
2023-07-03T01:38:30.505931 [ASN1   ] [E] [    0] [/home/dz/srsRAN/lib/src/asn1/rrc/paging.cc][715] Decoding failure.
2023-07-03T01:38:30.505931 [ASN1   ] [E] [    0] [/home/dz/srsRAN/lib/src/asn1/rrc/paging.cc][755] Decoding failure.
2023-07-03T01:38:30.505931 [RRC    ] [E] Failed to unpack PCCH message (7 B)
    0000: 48 1b 22 2a fc 41 83

And the nas pcap was  here:
[image]
[image]

Could you help to fix this problem, thanks in advance!

BRs
Lee

andrepuschmann, 2023-07-03:
Hey - could you please share the MAC PCAP that contains the paging message? Thanks

hz-nokia, 2023-07-04:
Hi andre,

All the msg in MAC PCAP were same which the info contained illegal checksum as following:
[image]

BRs
Lee

hz-nokia, 2023-07-05:
@andrepuschmann @djanderson @ttsou

hz-nokia, 2023-07-05:
Hi andre,
   I have uploaded the info in srsran github issue page.

BRs
Lee

andrepuschmann, 2023-07-05:
Could you please upload the PCAP, not a screenshot of Wireshark? Thanks

hz-nokia, 2023-07-06:
Hi andfre,

PCAP file was not supported when upload it. So I have add a postfix of txt in filename, you can change it to pcap when using.
I have uploaded 3 file which contained mac pcap, nas pcap and ue log as following:

[ue_mac_pcap.txt](https://github.com/srsran/srsRAN_4G/files/11963013/ue_mac_pcap.txt)
[ue_nas_pcap.txt](https://github.com/srsran/srsRAN_4G/files/11963014/ue_nas_pcap.txt)
[ue.log](https://github.com/srsran/srsRAN_4G/files/11963015/ue.log)