srsran/srsRAN_4G #1195: UE is not attaching to eNB

sruthi81, 2023-07-06:
## Issue Description ##
srsUE is not attaching to srsENB for 5G SA with ADALM Pluto hardware devices

## Setup Details ##
srsUE and srsENB are setup in two seperate machines and to each machine an ADALM Pluto is connected. Open5gs is used as the core network.
## Expected Behavior ##
UE attaches to eNB and ping transfer is possible

## Actual Behaviour ##

UE Console

[image]
[image]

eNB Console

[image]

## Steps to reproduce the problem ##
[ue_conf.txt](https://github.com/srsran/srsRAN_4G/files/11967761/ue_conf.txt)
[enb_conf.txt](https://github.com/srsran/srsRAN_4G/files/11967760/enb_conf.txt)

## Additional Information ##
[ue.log](https://github.com/srsran/srsRAN_4G/files/11967532/ue.log)
[enb.log](https://github.com/srsran/srsRAN_4G/files/11967712/enb.log)

andrepuschmann, 2023-07-06:
I am not sure this is even possible. The minimum bandwidth for SA is 5 MHz. Sample rate 5.76msps. I don't think the Pluto can even work with this rate over USB2.

sruthi81, 2023-07-07:
Ok Thankyou Sir.

closed 2023-07-07 by andrepuschmann

sruthi81, 2023-07-11:
i had a doubt the adalp pluto specifications stating that the max sample rate can go to 56 mhz and the band width upto 20 mhz. Then it should work for 5G?