srsran/srsRAN_4G #1205: Make Build failed!!

shanmukhapothana, 2023-07-20:
Hi Team,

We are try to deploying the srsRAN4G in Raspberry pi4b. and following the link is:

https://docs.srsran.com/projects/4g/en/next/app_notes/source/pi4/source/index.html

Following the below steps to build in Raspberry pi4b by using ubuntu 22.04 OS.

sudo apt update
sudo apt upgrade
sudo apt install cmake

sudo apt install libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev
git clone https://github.com/srsRAN/srsRAN_4G.git
cd srsRAN_4G
git checkout tags/release_19_12
mkdir build && cd build
cmake ../
make -j4
While running the make command, we are getting this below error.

sudo make -j4
Consolidate compiler generated dependencies of target srslte_agc
Consolidate compiler generated dependencies of target srslte_enb
[  0%] Built target srslte_agc
[  2%] Built target srslte_asn1
[  2%] Built target asn1_utils
[  2%] Built target srslte_enb
Consolidate compiler generated dependencies of target srslte_phy_common
Consolidate compiler generated dependencies of target srslte_ch_estimation
Consolidate compiler generated dependencies of target srslte_mimo
Consolidate compiler generated dependencies of target srslte_fec
[  2%] Building C object lib/src/phy/mimo/CMakeFiles/srslte_mimo.dir/precoding.c.o
[  4%] Built target srslte_ch_estimation
[  5%] Built target srslte_phy_common
[  5%] Building C object lib/src/phy/fec/CMakeFiles/srslte_fec.dir/turbodecoder.c.o
[  5%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/cfo.c.o
[  6%] Building C object lib/src/phy/phch/CMakeFiles/srslte_phch.dir/cqi.c.o
/home/mobiveil/srsRAN_4G/lib/src/phy/phch/cqi.c:121:85: error: argument 3 of type ‘uint8_t[64]’ {aka ‘unsigned char[64]’} with mismatched bound [-Werror=array-parameter=]
  121 | int srslte_cqi_value_pack(srslte_cqi_cfg_t* cfg, srslte_cqi_value_t* value, uint8_t buff[SRSLTE_CQI_MAX_BITS])
      |                                                                             ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/mobiveil/srsRAN_4G/lib/include/srslte/phy/phch/pucch_cfg.h:25,
                 from /home/mobiveil/srsRAN_4G/lib/include/srslte/phy/ch_estimation/refsignal_ul.h:35,
                 from /home/mobiveil/srsRAN_4G/lib/include/srslte/phy/ch_estimation/chest_ul.h:42,
                 from /home/mobiveil/srsRAN_4G/lib/include/srslte/srslte.h:50,
                 from /home/mobiveil/srsRAN_4G/lib/src/phy/phch/cqi.c:22:
/home/mobiveil/srsRAN_4G/lib/include/srslte/phy/phch/cqi.h:146:97: note: previously declared as ‘uint8_t *’ {aka ‘unsigned char *’}
  146 | SRSLTE_API int srslte_cqi_value_pack(srslte_cqi_cfg_t* cfg, srslte_cqi_value_t* value, uint8_t* buff);
      |                                                                                        ~~~~~~~~~^~~~
[  6%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/cp.c.o
/home/mobiveil/srsRAN_4G/lib/src/phy/mimo/precoding.c:787:47: error: argument 4 of type ‘float *[2]’ with mismatched bound [-Werror=array-parameter=]
  787 |                                        float* csi[SRSLTE_MAX_CODEWORDS],
      |                                        ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/mobiveil/srsRAN_4G/lib/src/phy/mimo/precoding.c:28:
/home/mobiveil/srsRAN_4G/lib/include/srslte/phy/mimo/precoding.h:92:58: note: previously declared as ‘float *[4]’
   92 |                                                   float* csi[SRSLTE_MAX_LAYERS],
      |                                                   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [lib/src/phy/phch/CMakeFiles/srslte_phch.dir/build.make:76: lib/src/phy/phch/CMakeFiles/srslte_phch.dir/cqi.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2706: lib/src/phy/phch/CMakeFiles/srslte_phch.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  6%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/find_sss.c.o
[  7%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/gen_sss.c.o
In file included from /home/mobiveil/srsRAN_4G/lib/src/phy/fec/turbodecoder.c:102:
/home/mobiveil/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h: In function ‘tdec_winarm16_alpha.isra’:
/home/mobiveil/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:315:44: error: ‘old[7]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  315 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/home/mobiveil/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:690:31: note: ‘old[7]’ was declared here
  690 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
[... 46 lines cut ...]
[  8%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/psss.c.o
cc1: all warnings being treated as errors
make[2]: *** [lib/src/phy/fec/CMakeFiles/srslte_fec.dir/build.make:216: lib/src/phy/fec/CMakeFiles/srslte_fec.dir/turbodecoder.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2420: lib/src/phy/fec/CMakeFiles/srslte_fec.dir/all] Error 2
[  8%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/refsignal_dl_sync.c.o
[  8%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/sfo.c.o
/home/mobiveil/srsRAN_4G/lib/src/phy/mimo/precoding.c: In function ‘srslte_predecoding_type’:
/home/mobiveil/srsRAN_4G/lib/src/phy/mimo/precoding.c:1925:16: error: ‘srslte_predecoding_diversity_multi’ accessing 32 bytes in a region of size 16 [-Werror=stringop-overflow=]
 1925 |         return srslte_predecoding_diversity_multi(y, h, x, csi, nof_rxant, nof_ports, nof_symbols, scaling);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/mobiveil/srsRAN_4G/lib/src/phy/mimo/precoding.c:1925:16: note: referencing argument 4 of type ‘float **’
/home/mobiveil/srsRAN_4G/lib/src/phy/mimo/precoding.c:784:5: note: in a call to function ‘srslte_predecoding_diversity_multi’
  784 | int srslte_predecoding_diversity_multi(cf_t*  y[SRSLTE_MAX_PORTS],
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[  9%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/sss.c.o
[  9%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/ssss.c.o
[  9%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/sync.c.o
[  9%] Building C object lib/src/phy/sync/CMakeFiles/srslte_sync.dir/sync_nbiot.c.o
[  9%] Built target srslte_sync
cc1: all warnings being treated as errors
make[2]: *** [lib/src/phy/mimo/CMakeFiles/srslte_mimo.dir/build.make:90: lib/src/phy/mimo/CMakeFiles/srslte_mimo.dir/precoding.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2602: lib/src/phy/mimo/CMakeFiles/srslte_mimo.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

Please help me to resolve this issue.

cloudwiser, 2023-07-20:
Are you running 32-bit or 64-bit Ubuntu 22.04 on the Pi4? It needs to be 64-bit.

shanmukhapothana, 2023-07-20:
Yes, I am using Ubuntu 22.04 64 bit OS only

shanmukhapothana, 2023-07-20:
yes,I am running on 64-bit Ubuntu 22.04 OS only.

Mobiveil INC., CONFIDENTIALITY NOTICE: This e-mail message, including any attachments, is for the sole use of the intended recipient(s) and may contain proprietary confidential or privileged information or otherwise be protected by law. Any unauthorized review, use, disclosure or distribution is prohibited. If you are not the intended recipient, please notify the sender and destroy all copies and the original message.

andrepuschmann, 2023-07-20:
This combination of build isn't really supported. Since you're using the srsRAN version in the appnote I'd also stick with the gcc shipped with the Ubuntu 20,04 LTS that was used back then.

shanmukhapothana, 2023-07-20:
Hi
Instead of following release cmd git checkout tags/release_19_12.Could I run latest release!!

shanmukhapothana, 2023-07-21:
Hi Team 
We are trying to deploying the srsRAN4G in Raspberry pi4b by using 64-bit Ubuntu 22.04.

Please suggest us whether we should follow old release or latest release.

If you have any other links, please share us.

andrepuschmann, 2023-07-21:
I don't have a definite advice as I've not run the eNB on the Pi4 for a long time. The latest version should build on Ubuntu 22.04 though. Why don't you just give it a try and report here?

andrepuschmann, 2023-07-21:
In any case - be advised that the 4G eNB on Raspberry Pi was and is a technical proof of concept. It worked and you can attach a UE but it's not something I would use for more serious tasks unless you are willing to spend time on optimization and tweaking.

shanmukhapothana, 2023-07-21:
After changing the latest version, build was successfully done.

Thanks for your advice and support.

closed 2023-07-21 by shanmukhapothana

andrepuschmann, 2023-07-21:
You're welcome