srsran/srsRAN_4G #1216: SRSUE SIGNAL FROM FILE SOURCE

ivyselly, 2023-08-23:
## Issue Description ##
Please i'd like some clarifications as to whether it possible to configure the SRSUE to capture and save RF signals from the usrp to a file.
Also is it possible to configure the SRS UE to read  signals from a file source and not the usrp?

## Setup Details ##
I'm using 
         linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.4.0,
         USRP B210, Ubuntu 22.04.3 LTS

## Additional Information ##
I see that UHD has some example python programs that can capture the files and store them to a file, however i can't locate the file after running the program and i am unable to edit the scripts.

I have been aple to generate some files using th GNU radio, but there'es still the issue as to whether i can use these files as a source to my srsue.

## Steps to reproduce the problem ##
Please let me know if this is possible at all, and necessary configs that i'd need to make, i appreciate all contributions and do let me know if i need to make any extra clarifications on this issue.

ivyselly, 2023-08-23:
@andrepuschmann  any help please?

andrepuschmann, 2023-08-30:
Hey @ivyselly - I am afraid reading and writing IQ samples from/to a file is not possible in srsUE. The UE is made for workign with real RF only. There are many test and example applications in the repo that can be used to capture samples for offline analysis.

closed 2023-08-30 by andrepuschmann

ivyselly, 2023-09-11:
thank you @andrepuschmann , is there any opplication that you can suggest for offline analysis of captured LTE signals please

mohmd-abzd, 2024-03-18:
Hi @ivyselly,

Have you come with a way to pull out the I/Q samples?