srsran/srsRAN_4G #1222: NR_CELL_SEARCH_TEST CRASHES WHEN RUN

ivyselly, 2023-09-06:
## Issue Description ##
I try running the /lib/srsRANRF/build/srsue/src/phy/test$ ./nr_sa_cell_search_test  but keep having the srsRAN crash, i have provided error output below.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
 
UBUNTU 20.04, usrob210

## Expected Behavior ##
[What you expect to happen]
i expected to get a list of available cells and their earfcns just like it provides for the cell_search test 

[image]

## Actual Behaviour ##
[What happens instead e.g. error message]
`./nr_sa_cell_search_test 

Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Opening 1 channels in RF device=default with args=default
Supported RF device list: UHD zmq file
Trying to open RF device 'UHD'
2023-09-06T16:48:26.637607 [RF     ] [I] Configuring physical DL channel 0 with band-pass filter (0.0, 0.0)
2023-09-06T16:48:26.637608 [RF     ] [I] Configuring physical UL channel 0 with band-pass filter (0.0, 0.0)
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.4.0.HEAD-0-g5fac246b
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened

Warning: TX gain was not set. Using open-loop power control (not working properly)

2023-09-06T16:48:28.147495 [RF     ] [I] Mapping RF channel 0 (device=0, channel=0) to logical carrier 0 on f_rx=3510.0 MHz
Setting manual TX/RX offset to 0 samples
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Operation not permitted
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Operation not permitted
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Operation not permitted
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Operation not permitted
2023-09-06T16:48:28.351211 [RF     ] [I] Mapping RF channel 0 (device=0, channel=0) to logical carrier 0 on f_rx=3510.0 MHz
2023-09-06T16:48:28.351212 [RF     ] [I] RF Rx channel 0 already on freq
2023-09-06T16:48:28.538383 [RF     ] [I] Mapping RF channel 0 (device=0, channel=0) to logical carrier 0 on f_rx=3510.0 MHz
2023-09-06T16:48:28.538387 [RF     ] [I] RF Rx channel 0 already on freq
2023-09-06T16:48:28.640377 [RF     ] [I] Mapping RF channel 0 (device=0, channel=0) to logical carrier 0 on f_rx=3510.0 MHz
2023-09-06T16:48:28.640378 [RF     ] [I] RF Rx channel 0 already on freq
2023-09-06T16:48:28.742385 [RF     ] [I] Mapping RF channel 0 (device=0, channel=0) to logical carrier 0 on f_rx=3510.0 MHz
2023-09-06T16:48:28.742386 [RF     ] [I] RF Rx channel 0 already on freq
2023-09-06T16:48:28.844417 [RF     ] [I] Mapping RF channel 0 (device=0, channel=0) to logical carrier 0 on f_rx=3510.0 MHz
2023-09-06T16:48:28.844417 [RF     ] [I] RF Rx channel 0 already on freq
terminate called without an active exception
--- command='./nr_sa_cell_search_test' version=23.04.0 signal=6 date='06/09/2023 16:48:29' ---
	./nr_sa_cell_search_test(+0x6ae9a) [0x564f5a017e9a]
	/lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f46e1e42520]
	/lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c) [0x7f46e1e96a7c]
	/lib/x86_64-linux-gnu/libc.so.6(raise+0x16) [0x7f46e1e42476]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0xd3) [0x7f46e1e287f3]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xa2b9e) [0x7f46e22a2b9e]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xae20c) [0x7f46e22ae20c]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xae277) [0x7f46e22ae277]
	/usr/local/lib/libsrsran_rf_uhd.so(+0x17239) [0x7f46e24d9239]
	/usr/local/lib/libsrsran_rf_uhd.so(+0x1856a) [0x7f46e24da56a]
	/lib/x86_64-linux-gnu/libc.so.6(__cxa_finalize+0xb6) [0x7f46e1e45a56]
	/usr/local/lib/libsrsran_rf_uhd.so(+0xc917) [0x7f46e24ce917]
srsRAN crashed. Please send this backtrace to the developers ...
---  exiting  ---
`
## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

run the  /lib/srsRANRF/build/srsue/src/phy/test$ ./nr_sa_cell_search_test

ANy feedback will be great!

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

FengShaocong, 2023-10-31:
hello, did you solve the problem? I also encountered this error.

Edm0ndDantes, 2025-03-25:
Hello, are there any updates on Tim's issue?