srsran/srsRAN_4G #1234: Multiple half-duplex SDRs instead of a single full-duplex SDR?

tomdionysus, 2023-10-20:
Apologies if this is the wrong place to ask, but is it possible (or, how hard would it likely be) to use two half-duplex SDRs (e.g. 2x HackRF One) instead of a full-duplex transceiver? My (likely naive) take is if we configure one for TX and one for RX, surely the eNB could just split the calls between the two drivers.  

I've looked into the Soapy driver and it doesn't seem to have problems running multiple devices, but before I dig into the code, are there dragons down there, is it worth exploring? If this is doable, this might help some others (#287, #1145) - and one reason, as ever, is cost. I can get 2x HackRF One units for less than USD $180, whereas a Nuand BladeRF costs over $540. 

Capability like this, if possible, would dramatically reduce the cost to people doing LTE PoCs with the srsRAN project.

JoshuaCurry, 2023-11-27:
I am very interested in this feature if it were possible to implement. HackRFs are great for the price point and also have a reference clock input which is much more useful compared to other devices like the LimeSDR Mini.
 
Other than having to implement some code that interfaces with Soapy that can read from one SDR and write to the other, is anyone aware of any technical reasons why this approach wouldn't work?

tomdionysus, 2023-11-29:
I've jumped the gun and started on a Soapy driver for running 2x HackRF One as a single full duplex device. This moves the project outside of the scope of srsRAN, as we'd just configure an unmodified srsenb to use the new driver.

I'm currently waiting on the hardware arriving to continue. If the srsRAN engineers have any further comments or advice it would be very much appreciated - i'll leave this issue open on my end and update it if I get it working.

Please note this is a pre-alpha draft: https://github.com/tomdionysus/SoapyHackRFDuplex

anoduck, 2024-03-09:
@tomdionysus This is exciting and fascinating news. I was just researching the possibilities of doing this when I discovered this issue.

tomdionysus, 2024-03-10:
@anoduck I got both modules working, but uncovered a deeper issue, please see the issue here: https://github.com/tomdionysus/SoapyHackRFDuplex/issues/1